roboticslab-uc3m/questions-and-answers #2: Major code reorganization among repositories

jgvictores, 2016-11-12:
For starters, I'd elaborate a list of all the functional libraries (these will mostly be yarp devices) that are of use, and functional programs. Once we lay the tree out, it will be easier to see what goes where. This could take some time!

jgvictores, 2016-11-25:
Started project: https://github.com/orgs/roboticslab-uc3m/projects/2

jgvictores, 2017-02-10:
Here some notes regarding what @PeterBowman and me have been  commenting.

## Common (OK)
- QA
- best-practices
- installation-guides
- openrave-yarp-plugins
- project-generator

## Common (:truck:)
- (migrate sf to gh) travislib
- teo-body (plugins:canBus,jr3)
- teo-head (programs:haar,color,asr)(plugin: tts(#38))
- asibot-main (https://github.com/roboticslab-uc3m/asibot-main/issues/21, https://github.com/roboticslab-uc3m/asibot-main/issues/28)
- teo-main (https://github.com/roboticslab-uc3m/teo-main/issues/74)

## Research (OK)
- textiles
- xgnitive

## TEO
- teo-openrave-models
- (must create) asibot-openrave-models
- follow-me

## Event (OK)
- workshop-humanoids2016

## Dont-touch
- manip-waiter (contains duplicates)(@jmgarciah)
- waiter (@jmgarciah)
- fusion-control (@jmgarciah)
- gait (@munozyanez)
- gaitcontrol (@munozyanez)
- TEO_push (TFM @j-lorente)
- teo_batc (TFM @RaulFdzbis)
- teo-ros-simulation (TFM @RaulFdzbis)
- teo-gazebo (TFM @RaulFdzbis)
- force-torque-balance (TFM @lpinel)
- teo_robot (TFM @RaulFdzbis)
- teotragen_com (@domingoesteban)
- TEOTraGen (@domingoesteban / @munozyanez)

## Forks
- open_abb (used for textiles/unfolding-industrial)
- LoliRepo (TFM @lpinel) (contains jr3 program, serves as base for teo-body jr3 plugin)

jgvictores, 2017-02-10:
[image]

David-Estevez, 2017-02-11:
OK means that I passed the exam? :sweat_smile:

jgvictores, 2017-02-11:
@David-Estevez LOL :smile:

jgvictores, 2017-02-17:
[image]

jgvictores, 2017-03-05:
Regarding TEO, documenting current state here: https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/README.md

Another important discussion on all this is: https://github.com/roboticslab-uc3m/QA/issues/15 I've added to https://github.com/orgs/roboticslab-uc3m/projects/2

PeterBowman, 2017-07-11:
Most of this is already done (not exactly as we planned at first, though), we are tracking remaining work at https://github.com/orgs/roboticslab-uc3m/projects/2.

jgvictores, 2018-01-05:
Ok, closing issue and tracking remaining work at https://github.com/orgs/roboticslab-uc3m/projects/2 !!

closed 2018-01-05 by jgvictores