roboticslab-uc3m/questions-and-answers #5: Add CONTRIBUTING.md

jgvictores, 2016-11-26:
Sample: https://github.com/roboticslab-uc3m/teo-main/blob/develop/CONTRIBUTING.md

jgvictores, 2016-11-26:
Also look at: http://discourse.ros.org/faq

jgvictores, 2016-11-26:
And: http://wiki.ros.org/Support

David-Estevez, 2016-11-26:
Those ROS guides look really nice...

jgvictores, 2016-11-27:
Yet another good reference: https://github.com/blog/2267-introducing-github-community-guidelines

jgvictores, 2016-12-17:
Added at 47a8cc6c26080608da6fca938e0ffbf41b469795, but must improve.

jgvictores, 2017-01-10:
Yet another reference: http://wiki.ros.org/action/info/BestPractices

PeterBowman, 2019-08-03:
> Yet another good reference: https://github.com/blog/2267-introducing-github-community-guidelines

GitHb has made available a handy wizard at Insights > Community to choose and upload a code of conduct statement.

PeterBowman, 2021-11-09:
> Sample: https://github.com/roboticslab-uc3m/teo-main/blob/develop/CONTRIBUTING.md

I believe all our CONTRIBUTING.md files have been synchronized with that one. Moreover, when this file is missing in a specific repo, the global one is pulled from [roboticslab-uc3m/.github](https://github.com/roboticslab-uc3m/.github/blob/9cb9df19a6acfbaa7f8fc042d76e3ab07eb80118/CONTRIBUTING.md). @jgvictores what next?

jgvictores, 2021-11-09:
IMHO:
1) Update (not mandatory, forget it too tedious): http://robots.uc3m.es/index.php/TEO -> https://apps.robots.uc3m.es/index.php/TEO (I'll open an issue regarding the validness of the certificate)
2) Close this issue ^^

PeterBowman, 2021-11-09:
OK, but I'd rather forget about the wiki altogether and only recommend a GitHub search. I believe the most relevant stuff has been already migrated (?).

PS certain links stopped working after you deleted the redirections I had prepared at https://github.com/roboticslab-uc3m/questions-and-answers/issues/91#issuecomment-805120205. For instance, see [TEO Cluster: Network, IP, MAC and PC specification list](http://robots.uc3m.es/gitbook-teo-developer-manual/network-information.html). The wiki is archived, hence no one can edit and fix that link.

jgvictores, 2021-11-09:
> OK, but I'd rather forget about the wiki altogether and only recommend a GitHub search. I believe the most relevant stuff has been already migrated (?).

Perfect!

> PS certain links stopped working after you deleted the redirections I had prepared at #91 (comment).

Sorry about that again! 

> For instance, see TEO Cluster: Network, IP, MAC and PC specification list. The wiki is archived, therefore no one can edit and fix that link.

Oh, that was https://github.com/roboticslab-uc3m/teo-developer-manual/issues/19 anyway ^^ Some day we'll find a decent mechanism for that.

PeterBowman, 2021-12-28:
Done as part of https://github.com/roboticslab-uc3m/questions-and-answers/issues/93#issuecomment-1001805412.

As a side note, whenever a repo lacks CONTRIBUTING.md, it is automatically pulled from https://github.com/roboticslab-uc3m/.github/. Also, it is possible to show the contents of a README.md file added there on the org's main page (https://github.com/roboticslab-uc3m).

closed 2021-12-28 by PeterBowman