roboticslab-uc3m/questions-and-answers #6: I2t parameter configuration in TechnosoftIpos drivers

PeterBowman, 2016-12-23:
The screenshots below illustrate how are TechnosoftIpos drivers configured in ASIBOT at this time:

[image]

[image]

I'll focus on the `I2t Over current` checkboxes and text inputs inside the `Protections` group on both windows. The latter is accessible from the `Drive Info` button in the upper center of the first screenshot.

* Which one of these values is actually taken into consideration by the driver? As `Drive Info` doesn't let us modify `I2t`, could we suppose that these values are pulled from the firmware?
* How is `I2t` calculated for TEO?
* Regarding TEO, does this value depend on the joint the driver is associated to? (i.e. shoulders and elbows should pick higher values than wrists)
* Did you configure `I2t` through EasySetUp/EasyMotionStudio or via CAN?

CC @roboticslab-uc3m/teo.
@edwinDOS please expand in case I missed something.

jgvictores, 2016-12-26:
On my behalf, I can only say we do not touch this via CAN. It is set via EasySetUp/EasyMotionStudio.

@smcdiaz @jmgarciah Any help here on the `I2t` doubts for ASIBOT above, please?

jmgarciah, 2016-12-30:
The driver i2t should not be modified. In our case, we calculate the robot TEO i2t. In fact, we calculate the i2t of each of the joints depending on the type of electric motor used.

In the next web link, in chapter 9, it is explained how to calculate several related parameters.

http://www.maxonmotor.it/medias/sys_master/root/8819755417630/maxon-Formelsammlung-e.pdf?attachment=true

Finally, all the protections of the drivers and of the joints are programmed through EasySetUp

@PeterBowman @jgvictores @smcdiaz

PeterBowman, 2017-07-11:
There is not much we can do ATM. Our main constraint are the drivers themselves since we hit the firmware protections (5A for 10 seconds) sooner than the motor ones. In short, whenever ASIBOT starts from a fully horizontal position, it blocks right away due to the firmware I2t and resumes after 30 seconds. While this happens, the reference position is constantly changing, thus the robot arm usually travels to the target at a considerable speed (almost instantaneously) once the protection has *cooled down*. If we didn't manage to exit the dangerous horizontal zone, we hit the I2t again and so on. On the contrary, the motor's I2t holds for several minutes before reaching a critical value; in this case, motor position is not preserved and the arm drops.

Note how the process described above is different from TEO: the most critical protection is the motor's I2t, hence the need of performing complex analysis and calculations. As said before, there is not much we can do in the case of ASIBOT apart from avoiding current-demanding paths. We'll want to implement a means of monitoring the I2t, but this task should be tracked at `yarp-devices`.

closed 2017-07-11 by PeterBowman

jgvictores, 2019-07-13:
> In the next web [link](http://www.maxonmotor.it/medias/sys_master/root/8819755417630/maxon-Formelsammlung-e.pdf?attachment=true), in chapter 9, it is explained how to calculate several related parameters.

The link is broken, but seems interesting. I believe it is https://www.scribd.com/document/336858216/maxon-Formelsammlung-e-pdf

@PeterBowman Any idea on a best practice to perma-link this?

PeterBowman, 2019-07-13:
@jgvictores sure: [336858216-maxon-Formelsammlung-e-pdf.pdf](https://github.com/roboticslab-uc3m/questions-and-answers/files/3388713/336858216-maxon-Formelsammlung-e-pdf.pdf).

jgvictores, 2019-07-13:
> @jgvictores sure: [336858216-maxon-Formelsammlung-e-pdf.pdf](https://github.com/roboticslab-uc3m/questions-and-answers/files/3388713/336858216-maxon-Formelsammlung-e-pdf.pdf).

@PeterBowman Cool, thanks!