roboticslab-uc3m/questions-and-answers #8: RoboticsLabUC3M Youtube Channel

rsantos88, 2017-01-12:
Alberto Jardón y Carlos me han pedido subir un vídeo que grabamos ayer de la demo waiter al canal de RoboticsLabUC3M. ¿Podrías @jgvictores facilitarme usuario y contraseña?
Gracias!!!

jgvictores, 2017-01-12:
Yo sólo tengo de mi canal... Eso me parece que lo lleva Edu.

jgvictores, 2017-01-12:
...y la cámara buena, Santi!

jgvictores, 2017-01-18:
@rsantos88 ¿Todo OK respecto a esto?

rsantos88, 2017-01-18:
@jgvictores Si! ya tenemos el [video](https://www.youtube.com/watch?v=mZ-5I0wXxmY) subido al canal.
Cierro issue.

closed 2017-01-18 by rsantos88

jgvictores, 2017-05-06:
Canales de interés:
- https://www.youtube.com/user/RoboticsLabUC3M
- https://www.youtube.com/user/jgvictores
- http://www.dailymotion.com/santimorante

PeterBowman, 2017-10-11:
> Canales de interés:

\+ https://www.youtube.com/channel/UCpi9pRtQPsNPPseycu-tpJQ (Robótica Asistencial UC3M)