roboticslab-uc3m/questions-and-answers #13: Where should datasets be uploaded?

jgvictores, 2017-01-27:
Here are some solutions (updated from https://github.com/roboticslab-uc3m/xgnitive/issues/23):
1. Zenodo: We chose this option in the mentioned issue, which generates DOIs and is popular in the machine learning community. An example from XGNITIVE: https://zenodo.org/record/168156#.WIt3FlwmRh5
2. ResearchGate: Not sure if it still generates DOIs.
3. Mendeley: a new player in this area.

PD: While we used to publish in https://sourceforge.net/projects/roboticslab/files/Datasets/, these are more modern and probably better solutions.

David-Estevez, 2017-01-28:
Is there any limitations on dataset size? 

For some datasets, such as the garment 3D scans this would be a key factor to settle for one solution.

jgvictores, 2017-01-29:
From the [Zenodo FAQ](https://zenodo.org/faq):
> We currently accept up to 50GB per dataset (you can have multiple datasets); there is no size limit on communities. However, we don't want to turn away larger use cases. If you would like to upload larger files, please contact us, and we will do our best to help you.

PeterBowman, 2021-12-03:
In the past, we used to upload those datasets to our RL-UC3M server. Isn't this an option nowadays?

jgvictores, 2021-12-06:
> In the past, we used to upload those datasets to our RL-UC3M server. Isn't this an option nowadays?

Not really. It required a certain permissions/access level to the server, which is not easy to maintain. Additionally, scalability is an issue (I got complaints on large traffic of our largest file, a Robot Devastation `.iso`). My current recommendations would be:

- Informal (work in progress, internal, etc): Google Drive within `@ing.uc3m.es` domain. It's easy and popular.
- Formal release: Zenodo. It's easy and popular.

I could extend on reasons pro/cons of these options and on others, but it would be pretty long and redundant with respect to the above. This is just documenting my most intuitive and updated conclusions.

jgvictores, 2025-08-17:
> Perhaps [Hugging Face](https://huggingface.co/)? See [1](https://huggingface.co/docs/hub/datasets-adding), [2](https://huggingface.co/docs/datasets/upload_dataset).

Nice catch!

Regarding Zenodo:
1. There's still the advantage regarding the DOI that it generates.
2. Specifically in the case of European projects, https://doi.org/10.5281/zenodo.14864392 was nicely back-linked to the official project https://cordis.europa.eu/project/id/952091/results through some OpenAIRE mechanism (I only pointed Zenodo to the project, the other way around was automagic).