roboticslab-uc3m/questions-and-answers #14: How should I document my project's dependencies?

David-Estevez, 2017-02-10:
I am currently documenting the dependencies of the [textiles](https://github.com/roboticslab-uc3m/textiles) project and I haven't been able to find any instructions about how to document a project within the [RoboticsLab organization](https://github.com/roboticslab-uc3m).

Textiles is a large project mixing C++ and Python, using non-stable package versions for some dependencies in order to get state of the art methods not available in stable releases. 

As any other project depends on it (yet)... should I include my dependencies / instructions in my repo only? should I use whatever method has been chosen for the main repos of this organization? where can I find that method?

jgvictores, 2017-11-29:
Along this topic, a nice table developed by @David-Estevez can be found [here](https://github.com/roboticslab-uc3m/yarp-devices/blob/9b92f99eb99d6de553dbf1ed748cfa5ccef65556/doc/yarp-devices-install.md).

PeterBowman, 2021-12-28:
@jgvictores any follow-up?

jgvictores, 2022-01-02:
> @jgvictores any follow-up?

I'm happy with how we currently document dependencies, see any example: https://github.com/roboticslab-uc3m/yarp-devices/blob/e3e45205ba11a9c3ab9ac5beebdac74b22f1d515/doc/yarp-devices-install.md#installation-from-source-code

jgvictores, 2022-01-02:
PS: Also see a more complex case: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/99389db08da34594e3de54490036c4caa68e967d/doc/openrave-yarp-plugins-install.md#openrave-yarp-plugins-installation-from-source-code

PeterBowman, 2022-01-02:
Thanks! I'm closing, then.

closed 2022-01-02 by PeterBowman