roboticslab-uc3m/questions-and-answers #15: C++ namespace convention

PeterBowman, 2017-03-03:
As work on #2 goes on, we'll find out that former TEO-specific code is intented to target a wider range of robot platforms, thus rendering the omnipresent `teo::` namespace quite inappropriate for common libraries. Proposed solutions:

* coin a new and catchy namespace only for this kind of libraries
* create an omnibus prefix for every instance of **R**obotics **L**ab code to which existing and future namespaces would be appended: `rl::`, `rl::teo::`, `rl::asibot::`, `rl::amor::`, etc.

PeterBowman, 2017-03-05:
A new, catch-all namespace was proposed by @jgvictores: `roboticslab::`. It might seem to be a bit lengthy; however, per [Google's C++ Style Guide - General Naming Rules](https://guiquanz.gitbooks.io/google-cc-style-guide/content/ebook/Naming.html):

> Give as descriptive a name as possible, within reason. Do not worry about saving horizontal space as it is far more important to make your code immediately understandable by a new reader. Do not use abbreviations that are ambiguous or unfamiliar to readers outside your project, and do not abbreviate by deleting letters within a word.

Although it doesn't concern namespaces explicitly (*Function names, variable names, and filenames [...]*), it might give a valuable pointer on how to approach this question. Remember that long namespaces can be shortened with the help of [C++ namespace aliases](http://en.cppreference.com/w/cpp/language/namespace_alias):

```c++
namespace rl = roboticslab;
namespace teo = roboticslab::teo;
```

More examples and some background: http://foonathan.net/blog/2016/01/26/namespace-alias.html.

jgvictores, 2017-05-20:
Added checklist above to keep track of repos affected/done.

rsantos88, 2017-05-30:
Done it! You can find the changes in develop branchs:
* [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/54cbdb151630ec56d7694462b9152cd575855870)
* [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/9f995df9a92bc9f7c1be4451038b7f4f96c8f660)
* [vision](https://github.com/roboticslab-uc3m/vision/commit/47fd24ae282ecd804e1912fb22ad784dfd32de39)
* [speech](https://github.com/roboticslab-uc3m/speech/commit/c14044cbbea1b26a7b7ec8976dcc80b4bee06a02)
* [ tools](https://github.com/roboticslab-uc3m/tools/commit/947a3873139f4607c40f7e792e616b862351e9d2)

PeterBowman, 2017-05-30:
Compilation was broken at kinematics-dynamics due to the gait repository not being prepared for the namespace switch. This was fixed at roboticslab-uc3m/gait@4fc2c91 and roboticslab-uc3m/kinematics-dynamics@bda0441. @munozyanez please keep in mind that any repository which uses gait's public interface (e.g. [gaitcontrol](https://github.com/roboticslab-uc3m/gaitcontrol/blob/6804900/.gitmodules)?) must be updated accordingly, or just keep pointing at the current local gait HEAD (avoid `git pull && git submodule update`) until that happens.

munozyanez, 2017-05-30:
Ok, thanks a lot for the job and the warning, @PeterBowman , i will keep in mind and fix the codes.

rsantos88, 2017-05-30:
thanks @PeterBowman

jgvictores, 2017-05-30:
Thanks @rsantos88 and @PeterBowman for this!

@rsantos88 I'm seeing we'll have some trouble merging your fork of https://github.com/roboticslab-uc3m/openrave-yarp-plugins. We'll work on it together. :smile:

jgvictores, 2017-06-01:
Added `Document at https://github.com/roboticslab-uc3m/best-practices` task above, befor closing issue.

jgvictores, 2017-06-12:
Done at 72fa3adf9147d64aa5dfd1780ae13b52cb3820a6.

closed 2017-06-12 by jgvictores

PeterBowman, 2017-07-11:
`yarp-devices` is still using the old namespace, but I guess we'll catch up later due to ongoing work (too many branches to synchronize everything at once).

rsantos88, 2017-11-06:
done it [here](https://github.com/roboticslab-uc3m/yarp-devices/pull/129)!

jgvictores, 2017-11-06:
Did on yarp-devices [here](https://github.com/roboticslab-uc3m/yarp-devices/pull/129) [thanks @rsantos88 !!]