roboticslab-uc3m/questions-and-answers #16: Remove deprecated YARP control-board interface methods

PeterBowman, 2017-04-21:
Several long-time deprecated methods will get deleted with the next YARP release (2.3.70). Some of these include:

* `yarp::dev::IPositionControl::setPositionMode()`
* `yarp::dev::IVelocityControl::setVelocityMode()`

Said methods were migrated upstream to the `yarp::dev::IControlMode` interface ([docs](http://www.yarp.it/classyarp_1_1dev_1_1IControlMode.html)). Affected repositories:

- [x] [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics) (done at roboticslab-uc3m/kinematics-dynamics@2e90f99)
- [ ] [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins)
- [ ] [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices)
- [ ] [manip-waiter](https://github.com/roboticslab-uc3m/manip-waiter)
- [ ] [asibot-main](https://github.com/roboticslab-uc3m/asibot-main)

jgvictores, 2017-04-22:
:scream::scream::scream:

PeterBowman, 2017-05-13:
I'm wondering what happened to `yarp::dev::IPositionDirect::setPositionDirectMode()` at YARP's `devel`. Since there is no `yarp::dev::IControlMode::setPositionDirectMode(int)` counterpart, how are we supposed to set this mode and use the PositionDirect interface?

jgvictores, 2017-05-13:
Looks like a pretty old issue asking for an `upstream` label, as it apparently never even made it into the [IControlMode.h](https://github.com/robotology/yarp/blob/03bc180cc3a98c7283b6a3a5bf7bc8bf0bc84929/src/libYARP_dev/include/yarp/dev/IControlMode.h) interface of `master`, where there is a [VOCAB_CM_POSITION_DIRECT](https://github.com/robotology/yarp/blob/03bc180cc3a98c7283b6a3a5bf7bc8bf0bc84929/src/libYARP_dev/include/yarp/dev/IControlMode.h#L124) but no actual method.

It would be good for us to recover the method, because it's what we use for the Technosoft IPos driver `PT Mode` (driver's modality where we push motor positions into a buffer).

PeterBowman, 2017-05-17:
Note to myself: start thinking about dropping *old* `IControlMode` methods in favour of the more generic `IControlMode2::(set|get)ControlMode(s)` with vocabs ([reference](http://www.yarp.it/classyarp_1_1dev_1_1IControlMode2.html)). In fact, this is how we should use now said modes, provided we upgrade our implementations to adopt `IControlMode2`.

PeterBowman, 2017-05-19:
> Note to myself: start thinking about dropping old IControlMode methods in favour of the more generic IControlMode2::(set|get)ControlMode(s) with vocabs (reference).

Example: roboticslab-uc3m/yarp-devices@3c4e80a.

PeterBowman, 2017-05-21:
All done but a few deprecated commands used in [manip-waiter/programs/manipWaiterExecManip/InCvPort.cpp](https://github.com/roboticslab-uc3m/manip-waiter/blob/30a5995/programs/manipWaiterExecManip/InCvPort.cpp). I was told not to bother with this (@jgvictores).

closed 2017-05-21 by PeterBowman

PeterBowman, 2017-05-25:
Confirming that `setPositionMode` et al. will be deprecated soon and eventually removed: robotology/yarp#1223. This is the deployment schedule for the next two releases (per https://github.com/robotology/yarp/projects):

* YARP v2.3.70 (2017/06/15)
* YARP v2.3.72 (2017/10/15)

jgvictores, 2017-05-26:
Reminder to self: check with @jmgarciah before anything breaks on https://github.com/roboticslab-uc3m/manip-waiter