roboticslab-uc3m/questions-and-answers #17: Schedule periodic Travis cron jobs on YARP's devel branch

PeterBowman, 2017-04-22:
Following the lines of [this comment at roboticslab-uc3m/teo-main#8](https://github.com/roboticslab-uc3m/teo-main/issues/8#issuecomment-296371019), and also with #7 (latest-YARP policy) and #16 (YARP deprecations striking out of nowhere) in mind, it might be worth running periodic builds for each repo - say, once a week? - that would fetch YARP's devel branch to keep up with any breaking changes being prepared for the next release. The reasoning behind this is simple: use Travis' mechanisms to warn us against upcoming trouble even if nobody is keeping an eye on YARP's development progress. Such issues will surely arise at stale projects (like asibot-main was until quite recently).

One may use the `TRAVIS_EVENT_TYPE` [default environment variable](https://docs.travis-ci.com/user/environment-variables/#Default-Environment-Variables) to differentiate between the usual *push* jobs and the *cron* ones.

jgvictores, 2017-04-23:
Sounds like a great idea!

For teo-main I activated a daily job on master via web. I understand this can be done via `.travis`? Could you insert the magic lines on any `.travis` so it's just copy+paste for me on the rest of the repos?

PeterBowman, 2017-04-23:
@jgvictores done at roboticslab-uc3m/kinematics-dynamics@4362843 and scheduled to run weekly on local develop branch. Please take a look at roboticslab-uc3m/kinematics-dynamics@677eeab, too.

PeterBowman, 2017-05-21:
YARP's devel already requires CMake +3.0, thus builds for yarp-devices are currently [erroring](https://travis-ci.org/roboticslab-uc3m/yarp-devices/builds/234534066). Pulling packages from ppa:kalakris/cmake is not an option since it stalled at 2.8.11. Looks like we need either to:

* compile CMake ourselves and, preferably, store it in Travis' cache, or
* upgread CI builds to Trusty

I'd like to remark that we no longer need to add a `sudo apt-get install cmake` and, in spite of [what official packages indicate](http://packages.ubuntu.com/trusty/cmake), the default CMake version installed is 3.2.2 at the time of writing. Proof at https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/234523759 ([L277](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/234523759#L277)), ([L307](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/234523759#L307)).

PeterBowman, 2017-07-12:
I deem this done, all repos listed in the issue's description have been enabled but:

* [follow-me](https://github.com/roboticslab-uc3m/follow-me) and [xgnitive](https://github.com/roboticslab-uc3m/xgnitive): I'm not touching these
* [teo-main](https://github.com/roboticslab-uc3m/teo-main) (same goes for [asibot-main](https://github.com/roboticslab-uc3m/asibot-main) once CI builds will be enabled): already built against latest YARP master on a daily basis, while YARP devel is tested by each of its sub-repos
* [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices): let's analyze this case at roboticslab-uc3m/yarp-devices#117 and roboticslab-uc3m/yarp-devices#118

closed 2017-07-12 by PeterBowman