roboticslab-uc3m/questions-and-answers #19: Upgrade with private repos!

jgvictores, 2017-05-17:
We've upgraded @roboticslab-uc3m to an organization plan with unlimited free private repositories and unlimited seats (renewable after 2 years).

@PeterBowman All is set to import https://github.com/jgvictores/amor-api to https://github.com/roboticslab-uc3m :smile:

David-Estevez, 2017-05-17:
@jgvictores  Can we move also horus & light-dome?

jgvictores, 2017-05-17:
@David-Estevez That would be great!

David-Estevez, 2017-05-17:
I'll do that ASAP :+1:

PeterBowman, 2017-05-17:
Glad to hear that, @jgvictores! Would you mind transferring ownership to our org? I don't have sufficient permissions - no access to the *Settings tab* on amor-api.

jgvictores, 2017-05-17:
Migrated `amor-api`. :smile:

jgvictores, 2017-06-05:
@David-Estevez @PeterBowman All OK to close this issue?

PeterBowman, 2017-07-05:
Ping @David-Estevez.

closed 2017-07-05 by David-Estevez