roboticslab-uc3m/questions-and-answers #21: Mark deprecated functions through macro?

jgvictores, 2017-06-05:
Use case: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/98 suggests an API break, implemented in https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/21ec3ea0c32a7811461d55264592a3aa7dbe9486

**Should the old function be left around for a while, while marking it as deprecated (as in YARP and OpenRAVE)?**

Here are my favourite to least favourite options:
1. Yes, use YARP's mechanism.
1. Re-implement something like YARP's mechanism. 
1. No, just kill and go on.
1. Use some non-portable gcc-specific marker.

More options/opinions?

PeterBowman, 2017-06-05:
I might sound a bit selfish, but let's consider that we are the only consumers of our own code. That is, we are in full control of development, shipment and usage of roboticslab-uc3m software, which in turn means that any API breakage would just impact specific repos at our GitHub organization and maybe a few user forks/independent projects around the same ecosystem. I'd expand on point 3 and say:

3b. No. Open an issue wherever the offending function is located at. Label it as *announcement*. Use the search bar to localize any call to said function across roboticslab-uc3m and list all affected repos. If necessary, elaborate a removal plan and detail any steps that need to be taken to perform a seamless migration to the new API. Ping the corresponding team or whoever could need more action on their end. Proceed gradually and, finally, kill the function.

We may take advantage of the fact that superbuild-like repos usually pull upstream code at its `master` branch. On the other hand, git submodules point at a specific point in the upstream's commit history. Both dependency management mechanisms enable us to proceed in stages and defer the final removal until everything is tied up. Within the past week, we've seen a similar situation at [#15](https://github.com/roboticslab-uc3m/QA/issues/15#issuecomment-304869333).

jgvictores, 2017-06-05:
In addition to what you say, maybe add a CD_WARNING inside the function?
I'm also thinking about a #warning, but would have to look into how this is implemented in YARP and/or OpenRAVE.

PeterBowman, 2017-06-05:
> In addition to what you say, maybe add a CD_WARNING inside the function?

You mean inside the old function, while it's still marked as deprecated? I don't see the point in adding it to its replacement function, that kind of warnings should strike the developer in the eyes during the compilation phase - too easy to overlook on runtime.

>  I'm also thinking about a #warning, but would have to look into how this is implemented in YARP and/or OpenRAVE.

See [robotology/yarp/cmake/template/yarp_conf_system.h.in#L106-L130](https://github.com/robotology/yarp/blob/44a555c/cmake/template/yarp_conf_system.h.in#L106-L130). We could then add a `return false` in `ICartesianSolver::getNumLinks`, perhaps preceded by `CD_ERROR` with the corresponding notice.

jgvictores, 2017-06-05:
If we did something like [robotology/yarp/cmake/template/yarp_conf_system.h.in#L106-L130](https://github.com/robotology/yarp/blob/44a555c/cmake/template/yarp_conf_system.h.in#L106-L130), where would it go? ColorDebug? Some new macro repository?

PeterBowman, 2017-06-05:
> Some new macro repository?

Almost like killing a fly with an elephant gun :smile: (or is it a BFG?). It would feel out of place inside ColorDebug, though. I don't have an answer for the general case (yet), but I recall seeing preprocessor directives in there that *maybe* could benefit from replicating YARP's mechanism (something along the lines of `kinematics-dynamics-conf.h.in`?). More precisely, I'd see `_USE_LMA` ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/f359a10/libraries/TeoYarp/KdlSolver/ICartesianSolverImpl.cpp#L85)) getting handled by the `#cmakedefine` magic. The problem is that, in order to properly mark deprecations in the **public API**, we'd have to distribute (i.e. install) the configured header file as well. If we set aside the `_USE_LMA` thing for a moment, another option is copy-pasting those `#define`s *inside* `ICartesianSolver.h`. Nasty.

CMake seems to offer an automated way of generating export-like macros, including deprecation attributes: [GenerateExportHeader](https://cmake.org/cmake/help/v3.0/module/GenerateExportHeader.html). Mentioning for completeness, I haven't tested this yet.

PeterBowman, 2017-06-06:
The following comments arose from a quick chat with @jgvictores:

* GenerateExportHeader looks nice, we could generate one such header file per repo, name it after the repo itself and distribute along with the API headers. This is an example of invocation at [amor-api](https://github.com/roboticslab-uc3m/amor-api/blob/aaa883a/libraries/amor/CMakeLists.txt#L59-L61) adjusted for its particular needs:
  ```cmake
  generate_export_header(amor_api
                         BASE_NAME amor
                         EXPORT_FILE_NAME ${CMAKE_BINARY_DIR}/generated_include/amor_export.h)
  ```
* We'd rather put a `CD_WARNING` prior to returning from the deprecated function (compare with my [previous comment](https://github.com/roboticslab-uc3m/QA/issues/21#issuecomment-306281162)). The CD macro would point users at the correct function on runtime.

PS GenerateExportHeader is available since CMake 2.8.6 ([compatibility matrix](https://cmake.org/Wiki/CMake_Version_Compatibility_Matrix/StandardCMakeModulesFindL)).

jgvictores, 2017-06-13:
In the context of [the original issue](https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/98):
> * GenerateExportHeader looks nice, we could generate one such header file per repo, name it after the repo itself and distribute along with the API headers.

This has been implemented in https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/e05909832863329bbe09b436d32302210ef25072 and https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/fb7480ad423e60bb48c719f7ca7e88bd376edaf0. Note that the macro in the abstract base class. Accessing with some code like:
```c++
    int i;
    ICartesianSolver* ii;
    ii->getNumLinks(&i);
```
actually does the trick and gives a nice compiler deprecated warning:
```
/home/yo/repos/kinematics-dynamics/libraries/TeoYarp/KdlSolver/ICartesianSolverImpl.cpp: In member function ‘virtual bool roboticslab::KdlSolver::getNumJoints(int*)’:
/home/yo/repos/kinematics-dynamics/libraries/TeoYarp/KdlSolver/ICartesianSolverImpl.cpp:18:23: warning: ‘virtual bool roboticslab::ICartesianSolver::getNumLinks(int*)’ is deprecated (declared at /home/yo/repos/kinematics-dynamics/libraries/TeoYarp/ICartesianSolver.h:27) [-Wdeprecated-declarations]
     ii->getNumLinks(&i);
                       ^
```

> * We'd rather put a `CD_WARNING` prior to returning from the deprecated function (compare with my [previous comment](https://github.com/roboticslab-uc3m/QA/issues/21#issuecomment-306281162)). The CD macro would point users at the correct function on runtime.

This has been implemented in https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/a167107d1edee81dc25e310c66c7a072920b1e59.

jgvictores, 2017-06-13:
Yet another issue to be closed once documented at https://github.com/roboticslab-uc3m/best-practices :smile:

PeterBowman, 2017-06-13:
* [CMakeLists.txt](https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/f359a10...fb7480a#diff-af3b638bc2a3e6c650974192a53c7291): please `install(FILES)` the new header; see first use case at [GenerateExportHeader docs](https://cmake.org/cmake/help/v3.0/module/GenerateExportHeader.html) and [libraries/TeoYarp/CMakeLists.txt](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/0d9d173/libraries/TeoYarp/CMakeLists.txt#L33-L36).
* [libraries/TeoYarp/ICartesianSolver.h](https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/f359a10...fb7480a#diff-9838b7079cd5c90e99d02adfdaad880dR27): I'd go the YARP way and provide a common implementation such as:
  ```c++
  ROBOTICSLAB_KINEMATICS_DYNAMICS_DEPRECATED virtual bool getNumLinks(int* numLinks) = 0
  {
      CD_WARNING("getNumLinks deprecated: use getNumJoints instead.\n");
      return false;
  }
  ```
  This is mainly aimed at respecting the [DRY](https://en.wikipedia.org/wiki/Don%27t_repeat_yourself) rule. Keep in mind that [these lines](https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/f359a10...fb7480a#diff-66c8e9a3b42827408d40269bf334bc34R7) would be needlessly replicated in every `ICartesianSolver` implementation.

jgvictores, 2017-06-14:
1. Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/7d221ec3bc0e0f35b3366d8415a3c94ac5fe8214.

jgvictores, 2017-06-15:
2. Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/2f49edee876a39ee8992cd0485fe837f6e6a50e9 and https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/1bb84a7dad3ab12c9b4d05e5c872ac76f20bb6b2.

jgvictores, 2017-06-15:
Documented on best-practices at https://github.com/roboticslab-uc3m/best-practices/commit/73dc3c2f9f435e425371a2009dba3d13c0223c06.

@PeterBowman @olayasturias Closing issue. Please comment if I've done anything terribly wrong. A pat on the back or some :beers: will suffice elsewise.

closed 2017-06-15 by jgvictores

PeterBowman, 2017-06-15:
Nice! In addition of merely marking a function as deprecated, I'd still recommend devs to open an *announcement*-like issue and actively assist in the migration to the new API.

> Open an issue wherever the offending function is located at. Label it as announcement. Use the search bar to localize any call to said function across roboticslab-uc3m and list all affected repos. [...]

A month-long removal stage is not bad, but it's not unusual to spot deprecated functions hanging around for *years* in popular APIs. I'd not be surprised if developers in our ecosystem, especially maintainers/users of those less common components, will eventually miss such occasional events; hence my proposal.

jgvictores, 2017-06-15:
Ok https://github.com/roboticslab-uc3m/best-practices/commit/1feae0b65a0c13232197750a46ceabf03e2aecd1!

PeterBowman, 2017-06-16:
[Google Translate FTW!](https://github.com/roboticslab-uc3m/best-practices/commit/1feae0b65a0c13232197750a46ceabf03e2aecd1#diff-f9f03ee04e6294dbb212243728eb39c2) :metal:

[image]

Just kidding. BTW it's curious to see how GitBook messes up the formatting (numbering + indentation, even if we take into account that `github/es` is missing a space before each `1.`). Compare: [github/en](https://github.com/roboticslab-uc3m/best-practices/blob/26767cd8dfc5e5cbf0347dd2915465a4c6438504/en/best-practices.md), [github/es](https://github.com/roboticslab-uc3m/best-practices/blob/26767cd8dfc5e5cbf0347dd2915465a4c6438504/es/best-practices.md), [gitbook/en](https://roboticslab-uc3m.gitbooks.io/best-practices/content/en/best-practices.html), [gitbook/es](https://roboticslab-uc3m.gitbooks.io/best-practices/content/es/best-practices.html).

jgvictores, 2017-06-19:
Hey, Thanks for pointing that out! I totally missed how Google Translator removed those spaces LOL. [gitbook/es](https://roboticslab-uc3m.gitbooks.io/best-practices/content/es/best-practices.html) looking better after https://github.com/roboticslab-uc3m/best-practices/commit/7bc92937f2988b59d56c0968757819e21839ac2a.

PeterBowman, 2017-08-29:
Regarding what a deprecated function should return (how should it behave), see my comment at roboticslab-uc3m/kinematics-dynamics@106aff2.

PeterBowman, 2018-01-25:
Marking a function as deprecated does not work quite well with SWIG. See roboticslab-uc3m/kinematics-dynamics#132 and #43.