roboticslab-uc3m/questions-and-answers #23: Move or rename yarp plugins directories?

jgvictores, 2017-06-12:
As spoken with @David-Estevez the current names (`TeoYarp`, etc) are misleading. Options:
1. Rename.
1. Move contents up one level.

Opinions?

David-Estevez, 2017-06-13:
Just to clarify, I was talking about the [yarp-devices repo](https://github.com/roboticslab-uc3m/yarp-devices). In case there are differences between the code in this folder and the code in the same-level, I find correct to place all the sources for the plugins in a folder to distinguish them from the rest of the code. But, instead of naming this folder TeoYarp, as it is no longer related to Teo, I would call the folder `yarp-devices' (or another descriptive name). 

If all the plugins are similar,  I think it is ok to put them on the same level.

jgvictores, 2017-06-13:
Correct. Also clarifying, this happens among several repositories, that contain a YARP plugins folder which was not correctly renamed during [the great transition](https://github.com/roboticslab-uc3m/QA/issues/2). For instance:
- [ ] https://github.com/roboticslab-uc3m/yarp-devices/tree/develop/libraries/BodyYarp
- [ ] https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/develop/libraries/TeoYarp
- [ ] https://github.com/roboticslab-uc3m/vision/tree/develop/libraries/VisionYarp (maybe this last one is not as bad)

PeterBowman, 2017-06-13:
I'd pick a catchy name and use it consistently across all repos. Somebody took a step ahead a while ago and created a [`yarpplugins`](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/67e2c2d/yarpplugins) directory at openrave-yarp-plugins. On the other hand, YARP itself distinguishes between carriers and devices, placing them in their own folder: [yarp/src/{carriers+devices}](https://github.com/robotology/yarp/tree/9b32448/src).

rsantos88, 2017-11-07:
https://github.com/roboticslab-uc3m/yarp-devices/tree/develop/libraries/BodyYarp -> (done it [here](https://github.com/roboticslab-uc3m/yarp-devices/pull/130))
https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/develop/libraries/TeoYarp -> (done it [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/125))
https://github.com/roboticslab-uc3m/vision/tree/develop/libraries/VisionYarp (done it [here](https://github.com/roboticslab-uc3m/vision/pull/71))

jgvictores, 2017-11-07:
- [ ] https://github.com/roboticslab-uc3m/tools

rsantos88, 2017-11-08:
>  https://github.com/roboticslab-uc3m/tools

done it [here](https://github.com/roboticslab-uc3m/tools/pull/8)!!

rsantos88, 2017-11-15:
@jgvictores if you think that all of the changes are correct and there is nothing more to change, please close this issue :)

jgvictores, 2017-11-15:
Yes, I also did this on [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/f6d6c3e039771fc4c5fe4985f0a502ea34c43e32/libraries), so I think we're good to go.

@rsantos88 Thanks!

jgvictores, 2017-11-15:
> On the other hand, YARP itself distinguishes between carriers and devices, placing them in their own folder: [yarp/src/{carriers+devices}](https://github.com/robotology/yarp/tree/9b32448/src).

Regarding this, I think we're okay to go for now, no existing carriers, no short-term plans, no long-term plans of creating many.

closed 2017-11-15 by jgvictores