roboticslab-uc3m/questions-and-answers #24: Bootstrapped YCM breaks automated builds

PeterBowman, 2017-07-05:
A nasty bug was introduced right before the v0.2.3 and v0.4.0 releases of YCM. It's still present at latest HEAD, for this reason all Travis builds in YCM-enabled repos have been erroring for several weeks: robotology/ycm#118. While this issue is especially noticeable with regard to our CI tools, regular users may experiment the same problems if they recently upgraded their local YCM copy. Current workarounds include:

* For users: rollback YCM to latest stable release (`git checkout tags/v0.2.2`).
* For CI builds: instruct YCM bootstrapper to check out the `v0.2.2` tag (see roboticslab-uc3m/kinematics-dynamics@6a69f5d).

The latter patch has been applied to root `CMakeLists.txt` at the following repos:

* openrave-yarp-plugins
* kinematics-dynamics
* asibot-hmi
* teo-main

I'd like to leave this issue open until upstream YCM `master` is fixed. However, we may consider leaving the `YCM_TAG` thing for good and simply keep updating the release version number to avoid dealing with sudden breakages (one would expect that master branch should be stable, which was not the case here).

PeterBowman, 2017-12-22:
Fixed in [YCM v0.6.0](https://github.com/robotology/ycm/releases/tag/v0.6.0).

PeterBowman, 2018-01-17:
> Fixed in YCM v0.6.0.

Fixed upstream, but we shouldn't upgrade YCM yet since latest releases require CMake 3.0+. I dropped some inline comments to warn about this: [ref](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/a2f39187fd324a975ee35e8c82646e7b7d57f6d5/CMakeLists.txt#L55-L59).

> However, we may consider leaving the YCM_TAG thing for good and simply keep updating the release version number to avoid dealing with sudden breakages (one would expect that master branch should be stable, which was not the case here).

I'm happy with this. Closing.

closed 2018-01-17 by PeterBowman

jgvictores, 2018-01-17:
Perfect!