roboticslab-uc3m/questions-and-answers #25: Consider using Zenodo

PeterBowman, 2017-07-18:
See https://zenodo.org and https://github.com/robotology/ycm + https://zenodo.org/record/809413.

Maybe members of @roboticslab-uc3m/xgnitive and @roboticslab-uc3m/textiles could find it useful?

jgvictores, 2017-07-18:
Related: https://github.com/roboticslab-uc3m/QA/issues/13
And we've actually been doing some movements in xgnitive,  see https://github.com/roboticslab-uc3m/xgnitive/issues/30

Did you have some other use in mind (such as software in addition to datasets)?

PeterBowman, 2017-07-18:
> Did you have some other use in mind (such as software in addition to datasets)?

Mainly software, but I'm not familiarized with these repos. As I see it, it looks like a more *official* (in the academic/research sense) hosting service than GitHub. Perhaps other repositories (kinematics, gait) might fit into Zenodo's use cases, too?

PeterBowman, 2018-01-17:
Marking as a duplicate of what @jgvictores pointed at two comments above this (https://github.com/roboticslab-uc3m/questions-and-answers/issues/25#issuecomment-316069504). It looks like I forgot to use the search bar and this issue is already discussed elsewhere, please reopen if deemed otherwise.

closed 2024-12-13 by PeterBowman