roboticslab-uc3m/questions-and-answers #27: How to resolve strange characters in yarp

rsantos88, 2017-09-11:
I don't remember how to solve this annoying problem when I connect with a yarp rpc port and I press the direction keys:
```bash
yarp rpc /tts/rpc:s
^[[C^[[C^[[C^[[B^[[A

```

David-Estevez, 2017-09-11:
Those strange characters are how the terminal represents the arrow keys (and other key combinations such as cntrl+key). They typically appear when the application you are using doesn't support using the arrows to navigate the history (or that combination of keys).

To summarize: yarp ports don't support command history navigation through arrows (as far as I remember, I might be wrong).

rsantos88, 2017-09-11:
Yes, I remember that @jgvictores knew some linux packet that when you install it allow command history navigation in yarp ports...

PeterBowman, 2017-09-11:
It actually worked for me on some system, at some point (maybe Windows? can't remember). After opening a RPC connection via terminal, I saw a `>>>` prompt starting on each line, then it was possible to browse the command history with arrow-up/-down and traverse left-right on the current line.

jgvictores, 2017-09-11:
Sounds like some `readline` magic. I recall some `READLINE` directives in yarp cmake.

rsantos88, 2017-09-11:
OK!! I solved it!! try to install `libedit-dev` packet to allow for commands auto-completion and recompile YARP. 
It works!! :smile:

David-Estevez, 2017-09-11:
Great!! :clap: :clap: :clap: :clap: 

Maybe we should put this info in the `installation-guides`...

PeterBowman, 2017-09-14:
> Maybe we should put this info in the installation-guides...

Done at roboticslab-uc3m/installation-guides@c17a122, closing as resolved and kudos to @rsantos88!

closed 2017-09-14 by PeterBowman