roboticslab-uc3m/questions-and-answers #28: How can I program... by demonstration?

jgvictores, 2017-09-13:
Yes! That is the question... how can I PbD!?

PeterBowman, 2017-09-13:
Something like [this](https://www.youtube.com/watch?v=ZcTwO2dpX8A)?

David-Estevez, 2017-09-13:
I see [this](http://tinyurl.com/y9qlkyzg) has extended to issues too... ;)

jgvictores, 2017-09-13:
Esentially, you'll want some gravity compensation to help out. That's `[gcmp]` off the `BasicCartesianControl` :
http://robots.uc3m.es/dox-kinematics-dynamics/group__BasicCartesianControl.html

Option 1: Manually grab some "waypoints" and then create a program. For instance, in the joint space, do some `yarp rpc /robotName/manipulatorName/rpc:i` to get joint positions you can later use in your program as in [this example](https://github.com/roboticslab-uc3m/yarp-devices/tree/73cbd201df69ee19662cdb26a83d898669834bcb/example/cpp/exampleRemoteControlboard) or similar.

Option 2: Use [yarpdatadumper](http://www.yarp.it/yarpdatadumper.html) ([related](http://www.yarp.it/yarpdatadumperAppGenerator.html)) to record the full trajectory. Then use stuff from our [tools](https://github.com/roboticslab-uc3m/tools) repository. Specifically, you'll want the [PlaybackThread](https://github.com/roboticslab-uc3m/tools/tree/a7be63dc53090a5ca4ed19dc078ab3823aac1be3/libraries/ToolsYarpPlugins/PlaybackThread). You can find an example of use at [examplePlaybackThread](https://github.com/roboticslab-uc3m/tools/tree/a7be63dc53090a5ca4ed19dc078ab3823aac1be3/example/cpp/examplePlaybackThread).

Note: There are several alternatives to these approaches, but these are kind of nice. [yarpmanager](http://www.yarp.it/yarpmanager.html) has some record/playback facilities, but I haven't really tried them. Additionally, [yarpdataplayer](http://www.yarp.it/yarpdataplayer.html) does some playback, but for me it's nice to have something like what we have in the previously mentioned [tools](https://github.com/roboticslab-uc3m/tools) repository because it is a small library that can be used independently and is not coupled with any graphical interface.

jgvictores, 2017-09-13:
@rsantos88 I'd enjoy some feedback on this! Good to close the issue? Maybe we should document this somewhere ([teo-software-manual](https://github.com/roboticslab-uc3m/teo-software-manual) is looking pretty abandoned)?

rsantos88, 2017-09-14:
@jgvictores yes, for me it's ok. You can close it if you want ;)

jgvictores, 2017-09-14:
Cool!

closed 2017-09-14 by jgvictores

rsantos88, 2017-09-21:
> Option 2: Use yarpdatadumper (related) to record the full trajectory. Then use stuff from our tools repository. Specifically, you'll want the PlaybackThread. You can find an example of use at examplePlaybackThread and its corresponding test.

An example to record the left arm trayectory of Teo: 
* Terminal 1: run **launchManipulation**
* Terminal 2: run **yarpdatadumper --name /log** (the file will be saved in /log directory)
* Terminal 3: **yarp connect /teo/leftArm/state:o /log**

the acquisition will start.
By pressing CTRL+C the acquisition is terminated.