roboticslab-uc3m/questions-and-answers #32: Rename this repo?

jgvictores, 2017-10-07:
Something among the lines of the following should increase visibility:
- questions-answers
- questions-and-answers

Opinions?

PeterBowman, 2017-10-07:
I'm not opposing, but isn't this visible enough?

[image]

jgvictores, 2017-10-07:
IMHO, initials can always be misleading, especially to non-native speakers. I couldn't find this in our current best-practices, but at least it was in the Google code style last time I checked. :-)

PeterBowman, 2017-10-07:
:+1: The good thing is, we can keep the redirect from the shorter name (I bet that people don't like to type too much). I'd choose `questions-and-answers` as its more clear to me.

jgvictores, 2017-10-07:
Thanks! Done!

closed 2017-10-07 by jgvictores