roboticslab-uc3m/questions-and-answers #33: Curarse en salud con TEO

rsantos88, 2017-10-19:
Después de haber hablado con @smcdiaz  y con @jmgarciah, y después de todas las consecuencia que supuso el cambio y la redistribución de repositorios, así como las consecuencias de posibles cambios con la actualización de software como puede ser YARP (ver [issue](https://github.com/roboticslab-uc3m/installation-guides/issues/24)) o un futuro cambio de distros de linux en los PCs de Teo (ver [issue ](https://github.com/roboticslab-uc3m/yarp-devices/issues/118)) se ha llegado a la conclusión de que son interesantes tomar las siguientes medidas:
* Actualizar la partición arrancable de Backup que hay instaladas en cada uno de los PCs de TEO (backups realizados a fecha 22/04/2016 (ver [issue](https://github.com/roboticslab-uc3m/yarp-devices/issues/43)) a una imagen actual
* Posible instalación de un banco de pruebas para cualquier tipo de cambio importante que requiera TEO

Yo creo que con realizar el primer punto sería suficiente por el momento, para tener el robot siempre operativo ante cualquier cambio. Decirme que os parece.

jgvictores, 2017-10-19:
1. Ok!
1. Entiendo que @munozyanez ya dispone de un banco de pruebas, que es lo que ha utilizado para probar https://github.com/roboticslab-uc3m/yarp-devices/issues/118 . @munozyanez ¿Puedes confirmar?

jmgarciah, 2017-10-19:
Respecto a la pregunta de @jgvictores , creo que el banco de pruebas de @munozyanez es para poder utilizar las antiguas hicoCAN sobre la version 16.04 de Ubuntu. Que quizas sea algo muy puntual.

El banco de pruebas que propone @rsantos88 es mas bien para trabajar sobre versiones que afecten directamente al funcionamiento de TEO, y que en un futuro puedan servir para actualizar a un nuevo Backup. Disponemos de un tercer PC para poder montarlo si quisieramos.

Aun así, estoy de acuerdo con las dos propuestas de @rsantos88

munozyanez, 2017-10-19:
En el banco de pruebas tenemos una hicocan con un driver modificado para funcionar sobre el último kernel de linux. Está funcionando, y envía y recibe mensajes de canbus, pero todavía está en pruebas.

munozyanez, 2017-10-19:
Tener un sistema completo independiente del que se utiliza para desarrollo (punto 2) me parece también interesante.
Estas alternativas de arranque podrían estar en donde os parezca (partición, disco, usb). A la hora de arrancar los pcs, es cuestión de elegir el sistema que te convenga (estable, desarrollo, demos, etc...).

munozyanez, 2017-10-19:
Por cierto, lo que yo llamo banco de pruebas es un pc blanco que está en mi mesa del labo. Este banco tiene una cpu de teo (la de repuesto), con una hicocan (rota, solo funciona un puerto) y una tarjeta ISCM4805/ISCM8005 (de las antiguas) con un motor y un encoder.
Ahora tiene un disco duro con debian 9, también se pueden añadir los discos que os parezca e instalar lo que os parezca (ubuntu 16, etc)

AlvaroMartinezR, 2017-10-20:
Me parece buena idea tener un back-up, mas ahora que tenía pensado incluir el middleware de ArmarX.

David-Estevez, 2017-10-20:
Yo estoy a favor de todo lo que sea probar cosas sin que Teo se rompa :smile:

RaulFdzbis, 2017-10-24:
Yo también, siempre es bueno tener backups al día.

Con respecto a lo del banco de pruebas suena bien para hacer pruebas sencillas, para pruebas más complejas supongo que habrá que seguir utilizando al teo original. 

En definitiva secundo lo de David, a favor de todo lo que sea probar cosas sin que Teo se rompa

rsantos88, 2017-10-26:
De acuerdo con lo comentado en la issue, he realizado backups recientes de cada uno de los ordenadores de Teo, subidos al repositorio [SVN de TEO](http://robots.uc3m.es/svn/TEOrepo/) a fecha 23/10/2017:
* http://robots.uc3m.es/svn/TEOrepo/backups/head/20171023/backup_vision_ubuntu_23_10_2017.fsa
* http://robots.uc3m.es/svn/TEOrepo/backups/locomotion/20171023/backup_locomotion_debian_23_10_2017.fsa
* http://robots.uc3m.es/svn/TEOrepo/backups/manipulation/20171023/backup_manipulation_debian_23_10_2017.fsa

También se han actualizado las particiones arrancables, que pueden ser seleccionadas desde el menú grub de cada PC:
[image]

jgvictores, 2017-10-26:
@rsantos88 Oleeeee, genial!! 💃💃💃

jgvictores, 2017-10-28:
This issue was moved to roboticslab-uc3m/teo-main#20

closed 2017-10-28 by jgvictores