roboticslab-uc3m/questions-and-answers #37: Replace apt-get -> by apt even on Travis?

jgvictores, 2017-11-24:
See https://github.com/roboticslab-uc3m/installation-guides/issues/22 for origin.
Most of the other organization repos have no apt-get except for on `.travis.yml`.
Question: Replace apt-get -> by apt even on Travis?

PeterBowman, 2017-11-26:
For automated tasks such as CI builds, I think `apt` provides little to no advantages over `apt-get`.

Per https://askubuntu.com/a/446484:

> `apt-get` may be considered as lower-level and "back-end", and support other APT-based tools. apt is **designed for end-users (human)** and its output may be changed between versions.

And https://itsfoss.com/apt-vs-apt-get-difference/:

> So with apt, you get all the necessary tools in one place. You won’t be lost under tons of command options. The main aim of apt is to provide an efficient way of handling package in a way “pleasant for **end users**”.

We shouldn't ever need these features; for instance, color output and the progress bar would be entirely useless on Travis. BTW Travis docs always use `apt-get` (and even recommend to suppress stdout output with `-qq`, see [ref](https://docs.travis-ci.com/user/installing-dependencies/#Installing-Packages-on-Standard-Infrastructure)).

jgvictores, 2017-11-27:
Ok, thanks a lot for those pointers, @PeterBowman !

Closing issue as `invalid` because it's the closest to a `not so good idea`, and do not want to saturate number of labels.

closed 2022-05-23 by PeterBowman