roboticslab-uc3m/questions-and-answers #40: Thoughts about path-planning within roboticslab-uc3m

jgvictores, 2018-01-06:
Thoughts about path-planning within [roboticslab-uc3m](https://github.com/roboticslab-uc3m), and considerations on creating a path-planning repository within the organization.

Considerations to be taken into account **before** blindly doing this:
1. Path-planning is not required for much of our research. 2D/3D path-planning is mostly required in the presence of obstacles, which is not our current general use case *within our current research* (this can change with time). Cases where we **are** doing some path-planning:
    1. [2D navigation] TIAGo navigation, via ROS navigation stack.
    1. [2D navigation] There should be some remanent code in [asibot-main](https://github.com/roboticslab-uc3m/asibot-main) ravebot or tasks, where OpenRAVE was used. Should be migrated to [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins) if it hasn't been done yet.
    1. [3D grasping] There has been some progress with TEO in OpenRAVE in https://github.com/roboticslab-uc3m/teo-grasp for grasping a bottle. This may remain independent or some day be merged into [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins).
1. We already have some trajectory generation in [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/), which is a good place for it (not exactly path-planning, does not take obstacles into account).

As seen, the above candidates already have their place. Therefore, my recommendations are the following:
1. Not creating a new path-planning repository within this organization, at least for now.
1. Use this issue to track path-planning developments within this organization.
1. If results are in Cartesian space, keep close integration with [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/), which treats Cartesian to joint space conversions. Therefore, new issues will potentially arise at [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/) too.

jgvictores, 2018-01-06:
cc: @PeterBowman @rsantos88

jgvictores, 2018-01-06:
> If results are in Cartesian space, keep close integration with kinematics-dynamics, which treats Cartesian to joint space conversions. Therefore, new issues will potentially arise at kinematics-dynamics too.

New related issues for keeping close integration: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134 and https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/135

jgvictores, 2018-01-06:
Regarding trajectories in joint space, just a reminder that we have some tools in the, well, [tools](https://github.com/roboticslab-uc3m/tools) repository. Namely, as commented [here](https://github.com/roboticslab-uc3m/questions-and-answers/issues/28#issuecomment-329166785):
> Specifically, you'll want the [PlaybackThread](https://github.com/roboticslab-uc3m/tools/tree/a7be63dc53090a5ca4ed19dc078ab3823aac1be3/libraries/ToolsYarpPlugins/PlaybackThread). You can find an example of use at [examplePlaybackThread](https://github.com/roboticslab-uc3m/tools/tree/a7be63dc53090a5ca4ed19dc078ab3823aac1be3/example/cpp/examplePlaybackThread) and its corresponding [test](https://github.com/roboticslab-uc3m/tools/blob/a7be63dc53090a5ca4ed19dc078ab3823aac1be3/tests/testPlaybackThread/testPlaybackThread.cpp).

jgvictores, 2018-07-12:
Related: https://github.com/ros-industrial-consortium/tesseract (refs: https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications and https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/155#issuecomment-404444963)

PeterBowman, 2018-10-22:
Not exactly path-planning, but kinda related: https://github.com/robotology/navigation.

jgvictores, 2018-11-25:
See also: https://github.com/strands-project/strands_navigation

jgvictores, 2018-12-21:
TODO: See https://github.com/roboticslab-uc3m/kinematics-dynamics#similar-and-related-projects, specifically https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/3859656e713d3bda9a10b13157a19b3a8f9a190e/README.md#path-planning-trajectory-generation-and-optimization

PeterBowman, 2021-11-09:
I think the demo developed by @elisabeth-ms has some bits of path planning. It's an DL-based object detection app for grabbing stuff with one of TEO's arms.

jgvictores, 2021-11-09:
> I think the demo developed by @elisabeth-ms has some bits of path planning. It's an DL-based object detection app for grabbing stuff with one of TEO's arms.

Cool, nice catch! I'm totally seeing some OMPL at https://github.com/elisabeth-ms/teo-sharon/blob/cfc3a62270e130d0f3a8a8418c18b1a901508bea/programs/TrajectoryGeneration/TrajectoryGeneration.hpp#L17-L24 in addition to the KDL and FCL code. Thanks!

PeterBowman, 2022-03-13:
See also this grasping demo featuring the iCub: https://github.com/robotology/community/discussions/573.