roboticslab-uc3m/questions-and-answers #41: Install gtest packages in Travis builds, remove embedded files from test/

PeterBowman, 2018-01-06:
Currently, it's a common practice in repos that perform unit testing with Travis CI to host a `gtest-x.x.x` directory within `test/`, see [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/3b3600e4268e9580365bf3fc71bd0f0fcb46658b/test). Proposal: remove those files and install Google's testing framework via `apt`:

* https://packages.ubuntu.com/trusty/libgtest-dev (**1.6.0!**)
* https://packages.ubuntu.com/xenial/libgtest-dev (1.7.0)

PeterBowman, 2018-01-17:
Added TODO list in description. @jgvictores:

- OK to fiddle with xgnitive?
- yarp-devices is a bit different since its test scripts are ignored by Travis. Anyway, if we are to delete the `gtest-*` directory, we need to make sure that the `libgtest-dev` package is available. I did not find it in the archive Debian stretch database, hence it's marked as blocked for now until we move on to Xenial/Squeeze.

jgvictores, 2018-01-17:
These months `xgnitive` is going to be a bit crazy, I'd wait till IROS 2018 deadline March 1 2018, ok?
I think tools is a great place to start; I'll be working on https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134 at kinematics-dynamics but you can also work there as I'll just be touching TrajectoryLib on a separate branch.

PeterBowman, 2018-01-18:
Funnily enough, `libgtest-dev` doesn't contain binaries we could use right away to build our unit tests. Per https://www.eriksmistad.no/getting-started-with-google-test-on-ubuntu/:

> Note that this package only install source files. You have to compile the code yourself to create the necessary library files. These source files should be located at /usr/src/gtest. Browse to this folder and use cmake to compile the library:

The cost of not embedding GoogleTest framework in every single repository is... having to replicate the build instructions in `.travis.yml` to cover the compilation of GoogleTest itself.

jgvictores, 2018-01-19:
Yes, the [package description](https://packages.ubuntu.com/xenial/libgtest-dev) states `Google's framework for writing C++ tests - header files`.

I've been reading through a [related bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=802587), where there was a proposal to take advantage of a gtest/gmock merge for a change... Only thing we'll see in the future for now is a name change from `libgtest-dev` to `googletest` but still no binaries in the [file list](https://packages.ubuntu.com/zesty/googletest).

Not sure what the best solution here is. I guess one practical solution would be to determine which method is faster, but that's only one of the possible criteria.

jgvictores, 2018-01-19:
More bugs issued on Ubuntu end up pointing to bugs issued on Debian:
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=726021
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=845717

jgvictores, 2018-01-19:
Okay, from [ref 1 from above](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=726021): 
> Because of the C++ "One Definition Rule", gtest must be  compiled with exactly the same flags as your C++ code under test.

That... could actually make sense.

David-Estevez, 2018-01-19:
You might already know this, but this is what [Google recomends](https://github.com/google/googletest/blob/master/googletest/README.md#incorporating-into-an-existing-cmake-project):

*  Download the GoogleTest source code manually and place it at a known location. This is the least flexible approach and can make it more difficult to use with continuous integration systems, etc.
*  Embed the GoogleTest source code as a direct copy in the main project's source tree. This is often the simplest approach, but is also the hardest to keep up to date. Some organizations may not permit this method.
*  Add GoogleTest as a git submodule or equivalent. This may not always be possible or appropriate. Git submodules, for example, have their own set of advantages and drawbacks.
*  Use CMake to download GoogleTest as part of the build's configure step. This is just a little more complex, but doesn't have the limitations of the other methods.

PeterBowman, 2018-01-19:
> Use CMake to download GoogleTest as part of the build's configure step. This is just a little more complex, but doesn't have the limitations of the other methods.

YCM could care about such complexity for us. Actually, I forgot that this was already an option (sorry!): https://github.com/roboticslab-uc3m/questions-and-answers/issues/1#issuecomment-284085241.

I'm going to update the title accordingly.

jgvictores, 2018-01-20:
Are we sure we do not prefer pulling the source via `apt`?
- Pro: looks easier
- Con: Older version, I guess

PeterBowman, 2018-01-20:
It would force us to build and compile it *by hand* (in Travis, build steps should be carefully replicated) instead of letting CMake do its work (current behavior). Also, the package solution implies compiling with `sudo`, manually copying/symlinking binaries, etc. ([example](https://www.eriksmistad.no/getting-started-with-google-test-on-ubuntu/)).

jgvictores, 2018-01-20:
The ([example](https://www.eriksmistad.no/getting-started-with-google-test-on-ubuntu/)) needs to compile with `sudo` and manually copy because it does an in-source build. I think we could do better.
Anyway, no personal preference. I guess since it's source and has to be compiled, it makes more sense to download the latest code anyway.

PeterBowman, 2018-01-20:
> it makes more sense to download the latest code anyway

I was thinking about making YCM pull & compile the exact same version we use right now (1.7.0).

jgvictores, 2018-01-20:
> I was thinking about making YCM pull & compile the exact same version we use right now (1.7.0).

:+1:

PeterBowman, 2018-01-20:
> It would force us to build and compile it by hand (in Travis, build steps should be carefully replicated) instead of letting CMake do its work (current behavior).

I'm still pondering on this, sorry. I'm definitely positive on using CMake somehow, although the YCM solution might be a bit too convoluted. Vanilla CMake already offers a [FindGTest.cmake module](https://cmake.org/cmake/help/v3.0/module/FindGTest.html), I'll probably test it soon.

jgvictores, 2018-01-20:
> I'm still pondering on this, sorry. I'm definitely positive on using CMake somehow, although the YCM solution might be a bit too convoluted. Vanilla CMake already offers a [FindGTest.cmake module](https://cmake.org/cmake/help/v3.0/module/FindGTest.html), I'll probably test it soon.

No worries! It's definitely a good idea to have thought it thoroughly, before propagating a solution throughout all repositories.

PeterBowman, 2018-01-21:
FindGTest assumes that Google Test framework is already compiled, i.e. it looks for `libgtest.so` and `libgtest_main.so` (or `.a`) and reports an error if not found.

> Because of the C++ "One Definition Rule", gtest must be compiled with exactly the same flags as your C++ code under test.

No good, then. Another option: install package and `add_subdirectory(/usr/src/gtest)`.

jgvictores, 2018-01-21:
> Another option: install package and add_subdirectory(/usr/src/gtest)

Sounds okay. As long as we do an out-of-source build, we won't need the `sudo` for make.

PeterBowman, 2018-01-21:
@jgvictores, @David-Estevez I followed the "package + vanilla CMake" solution: roboticslab-uc3m/kinematics-dynamics@f2ae4ed. What do you think? Some points to consider:

* Refactor and place the CMake code in a new find module, say `FindGTestSources.cmake`. This is different from standard `FindGTest.cmake` in the way that the latter looks for compiled binaries, whereas our module would determine the location of headers and sources (for use as input in `add_subdirectory()`).
* Said find module could borrow some ideas from the *official* `FindGTest.cmake`, such as support within CTest. Feels like reinventing the wheel, though.
* I tested that commit with googletest both as an `apt` package and cloned via git. The latter is especially relevant on Windows (users may set the `GTEST_ROOT` environment variable to point at the appropriate location). Should work nicely with googletest 1.6 and 1.7.
* It should be noted that googletest 1.8 changed a lot and new paths have to be considered in the find module to work properly. Affects Ubuntu packages from Zesty (17.04) onwards.
* Provide installation instructions for `libgtest-dev` in our guides.

As a side note: perhaps `tests` would be a better choice than `test` for the directory name of unit tests? We already have `programs` and `libraries` (but: `cmake/template`). I recall seeing a top-level `src` somewhere. In fact, directory layout might be better covered in a new issue in best-practices.

PeterBowman, 2018-01-21:
Note to self: [Establish gtest version (SO)](https://stackoverflow.com/questions/14687282/establish-gtest-version).

David-Estevez, 2018-01-22:
> Edit: since a new dependency would be required to build unit tests (googletest package, no longer local), should they be disabled by default?

What about disabling them if googletest cannot be found?  (as devices do in yarp-devices if dependency is not found)

PeterBowman, 2018-01-22:
I'm not sure about this. You'll usually want to build tests in Travis and abort them if not successful, other scenarios are non-critical. Silently hiding an unsatisfied dependency or even just warning about it is perhaps a bit too risky in this case.

PeterBowman, 2018-02-05:
I kinda like it. @jgvictores, @David-Estevez what do you think?

* [`FindGTestSources.cmake`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/563556d0e50a2c71128dd6db8e4fd85f49066830/cmake/FindGTestSources.cmake) (supports googletest 1.6, 1.7, 1.8)
* usage in [`test/CMakeLists.txt`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/563556d0e50a2c71128dd6db8e4fd85f49066830/test/CMakeLists.txt#L6-L14) (required 1.6)
* [Travis build](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/337582054)

> As a side note: perhaps tests would be a better choice than test for the directory name of unit tests? We already have programs and libraries (but: cmake/template). I recall seeing a top-level src somewhere. In fact, directory layout might be better covered in a new issue in best-practices.

See #46.

> Edit: since a new dependency would be required to build unit tests (googletest package, no longer local), should they be disabled by default?

> You'll usually want to build tests in Travis and abort them if not successful, other scenarios are non-critical. Silently hiding an unsatisfied dependency or even just warning about it is perhaps a bit too risky in this case.

New issue?

> What about disabling them if googletest cannot be found? (as devices do in yarp-devices if dependency is not found)

See #18.

jgvictores, 2018-02-06:
IMHO, everything is fine. I'd even go for a solution like #18. Travis will fail with pretty obvious messages when attempting to run tests which have not been compiled if we ever forget to yaml gtest.

PeterBowman, 2018-02-06:
> I'd even go for a solution like #18. Travis will fail with pretty obvious messages when attempting to run tests which have not been compiled if we ever forget to ~~yaml~~YCM gtest.

Title updated, this is no longer a YCM issue according to the latest proposal.

I'm not sure about applying #18 here. We may forget to install libgtest-dev, but that should happen at most *once*. The question is: are unit tests valuable/relevant outside Travis CI? Should regular users care, and see annoying warnings about a missing package which would led them to dig into the installation guides?

jgvictores, 2018-02-07:
> Title updated, this is no longer a YCM issue according to the latest proposal.

Yes, I was about to do that, just couldn't think of an accurate description.

> I'm not sure about applying #18 here. We may forget to install libgtest-dev, but that should happen at most once. The question is: are unit tests valuable/relevant outside Travis CI? Should regular [...]

IMHO, the mechanism proposed at #18 only warns once, which isn't annoying. We are all developers and I think it's good for us to be informed if tests are not going to be compiled on our machine and why.
Workflow: get tired of warnings -> install gtest -> have tests compiled which at least checks if we break an API.

> PS sorry for "dizzying the partridge"!

Big LOL!

PeterBowman, 2018-02-09:
Done at kinematics-dynamics, added install guide:

* roboticslab-uc3m/kinematics-dynamics@93b37d4
* roboticslab-uc3m/installation-guides@1878368

PeterBowman, 2018-02-09:
> These months xgnitive is going to be a bit crazy, I'd wait till IROS 2018 deadline March 1 2018, ok?

@jgvictores OK! All other tasks are done, I leave xgnitive up to you. Closing issue.

closed 2018-02-09 by PeterBowman

PeterBowman, 2018-03-18:
XGNITIVE done at https://github.com/roboticslab-uc3m/xgnitive/compare/5d81fbc...06506e2.

jgvictores, 2018-03-18:
Thanks a lot!!

PeterBowman, 2018-04-03:
Note that googletest 1.8.0 forces installation of headers and generated binaries (https://github.com/google/googletest/commit/98d988deac06637364f6cd41c45c3db4a8a0b6bc). This behaviour may be suppressed on current googletest `master` (1.9.0?) with the `INSTALL_GTEST` CMake option (https://github.com/google/googletest/commit/0e8e0e07d6c4bc8c9cd6df5407452c12752ab45c), although it won't affect us since we don't process the root CMakeLists.txt where this option is defaulted to `ON` (`FindGTestSources.cmake` skips this directory and looks for `googletest/CMakeLists.txt`).

PeterBowman, 2019-02-27:
Alternatively, use [FetchContent](https://cmake.org/cmake/help/v3.11/module/FetchContent.html) module from CMake 11. This allows to fetch external projects on configure time. Added in YCM 0.10 (not released yet).

PeterBowman, 2019-04-19:
> Note that googletest 1.8.0 forces installation of headers and generated binaries ([google/googletest@98d988d](https://github.com/google/googletest/commit/98d988deac06637364f6cd41c45c3db4a8a0b6bc)). This behaviour may be suppressed on current googletest `master` (1.9.0?) with the `INSTALL_GTEST` CMake option ([google/googletest@0e8e0e0](https://github.com/google/googletest/commit/0e8e0e07d6c4bc8c9cd6df5407452c12752ab45c)), although it won't affect us since we don't process the root CMakeLists.txt where this option is defaulted to `ON` (`FindGTestSources.cmake` skips this directory and looks for `googletest/CMakeLists.txt`).

Things are getting weird on cosmic (18.10) and disco (19.04), the libgtest-dev package is no longer a mere placeholder for googletest (bionic, although it still depends on it) and now contains binaries along with gtest headers.

> Alternatively, use [FetchContent](https://cmake.org/cmake/help/v3.11/module/FetchContent.html) module from CMake 11. This allows to fetch external projects on configure time. Added in YCM 0.10 (not released yet).

This is currently broken, a fix is expected for YCM v0.10.2: https://github.com/robotology/ycm/pull/247.