roboticslab-uc3m/questions-and-answers #42: [ci] Compile examples

jgvictores, 2018-01-16:
Many repos have examples that are compiled separately... that should not mean not having them tested via Travis!

PeterBowman, 2018-01-16:
We could have avoided https://github.com/roboticslab-uc3m/tools/issues/10#issuecomment-357375402 if examples were covered by tests.

Also, keep this in mind: roboticslab-uc3m/best-practices#21. Which boils down to: place Travis instructions that relate to building the examples in the `before_script` YAML section.

PeterBowman, 2018-01-17:
Added TODO list to description.

PeterBowman, 2018-02-22:
CMake [alias targets](https://cmake.org/cmake/help/v3.0/manual/cmake-buildsystem.7.html#alias-targets) may be useful here.

PeterBowman, 2018-04-10:
Several examples are not even written in C/C++. Perhaps we could test bindings, too.

PeterBowman, 2019-06-11:
Idea: `add_subdirectory(examples)` from root *CMakeLists.txt*, set `ENABLE_examples=ON` in Travis (defaults to OFF)?

jgvictores, 2019-06-14:
IMHO:
1. CI is great, and it would be great to have `examples` at least compiled via Travis.
2. I'm happy with self-contained `examples`. If we `add_subdirectory(examples)`, wouldn't it be catching repo-level dependencies rather than forcing them to be cited explicitly?

As a solution, maybe forcing Travis to enter `examples` and compile separately?

PeterBowman, 2019-06-14:
ASWJ:

* Examples are going to remain as self-contained projects, no matter what.
* We could instruct root CMakeLists.txt to traverse these project so that everything is built together (main project + examples). However, this simple trick would force us to hide a few lines behind an `if` clause, for instance [these `find_package() instructions`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/3d2159b808dfda5737283390ee4b426e1ac27fe8/examples/cpp/exampleScrewTheoryTrajectory/CMakeLists.txt#L6-L9). Since we aim to illustrate best practices and usage of our libraries/APIs (targeted at devs), such hacks would be certainly misleading.
* Make Travis configure and compile the examples separately in the `before_script` section of .travis.yml.

closed 2019-06-16 by PeterBowman

jgvictores, 2019-07-10:
@PeterBowman Thanks!!!