roboticslab-uc3m/questions-and-answers #45: Install our YARP plugins elsewhere

jgvictores, 2018-02-02:
Install YARP plugins elsewhere: right now, in Ubuntu, our YARP plugins are installed in `/usr/local/lib/yarp`. However, icub plugins are installed in `/usr/local/lib/icub`.

PeterBowman, 2018-03-23:
Example: https://github.com/roboticslab-uc3m/tools/commit/55ab32f776022806be29fe518e6684285deca71e.

PeterBowman, 2018-03-23:
Done at:

* https://github.com/roboticslab-uc3m/tools/commit/55ab32f776022806be29fe518e6684285deca71e
* https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/2bf8ca208805c0044d3df7665eeddb8ebcbe249b
* https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/4b4ff60990cea60236b268d3c8f831fdb6542576
* https://github.com/roboticslab-uc3m/yarp-devices/commit/3db8791339f62f0e0e0989e901d24d61d85311f4
* https://github.com/roboticslab-uc3m/speech/commit/74a68d01d268802e4652f8bdbb5397b08f74663a

closed 2018-03-23 by PeterBowman

PeterBowman, 2018-03-24:
@roboticslab-uc3m/teo

It is advised to uninstall, pull and install again the projects listed in my previous comment, in case you are working with any of them. Alternatively, look for installed devices and their associated .ini descriptors at `/usr/local/lib/yarp` and `/usr/local/share/yarp/plugins`, respectively, and remove them "by hand". YARP's module loader will give priority to modules installed in the old path (common for its own devices, e.g. opencv_grabber and the like). You may use the `yarpdev --list` utility to check where your modules are located at (as seen by YARP) and apply this fix if needed.