roboticslab-uc3m/questions-and-answers #46: Singular or plural and other conventions

jgvictores, 2018-02-03:
Singular or plural and other conventions:
1. Derived from the constant `example` (yarp-ish) vs `examples` directory.
1. @PeterBowman pointed out this also happens with `cmake/template` vs `cmake/templates`

Proposed ToDo:
- [ ] Decide on standard.
- [ ] Update https://github.com/roboticslab-uc3m/project-generator (which BTW has no examples)
- [ ] Somewhere in https://github.com/roboticslab-uc3m/project-generator point to solution implemented in project-generator, as a recommendation to follow even when generating folder structure manually.

PeterBowman, 2018-02-03:
+1 for plural.

jgvictores, 2018-02-03:
> +1 for plural.

Yes, I think this is more intuitive.

PeterBowman, 2018-02-03:
Also, I started cleaning a bit the contents of `cmake/`, thus we may encounter subdirs such as:

* [cmake/find-modules/](https://github.com/roboticslab-uc3m/yarp-devices/tree/6dcce3b235dab788bda1b3425207b2d55b6cb074/cmake) (`FindPkg.cmake` and similar)
* [cmake/modules/](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/54c2c9fa5966ac36b0d17dfd080b70a0ba9be1bf/cmake) (general CMake utilities)

In the future, YCM build modules could be moved into `cmake/build-modules/`. I borrowed this idea from the [YCM repo](https://github.com/robotology/ycm/tree/331f21737df560e9a484c4ceb4f1398fe8ca2e14).

PeterBowman, 2018-02-03:
> In the future, YCM build modules could be moved into cmake/build-modules/. I borrowed this idea from the YCM repo.

I'd suggest another name since `.gitignore` will probably mess up with anything that starts with `build` ([proof](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/618857f0ab7297dec079bef0417b04ef1e1020e0/.gitignore#L8)). Perhaps... `cmake/ycm-modules/`, and also move `IncludeUrl.cmake` and `YCMBootstrap.cmake` in there.

jgvictores, 2018-02-05:
Updated description with `test`(s) from #41.

PeterBowman, 2018-02-06:
> Decide on standard.

Perhaps we could go a bit further and outline the standard directory layout for most projects, that is, the tree structure of folders (`libraries/`, `programs/`, `test()s/`, etc.). Yeah, this is somehow reflected in project-generator, but might be best doing it in a more official way with a mention in [developer-manual](https://github.com/roboticslab-uc3m/developer-manual).

jgvictores, 2018-02-06:
>  (...) might be best doing it in a more official way with a mention in [developer-manual](https://github.com/roboticslab-uc3m/developer-manual).

LGTM! Some comments:
1. Yes, I like the idea of a reference guide in [developer-manual](https://github.com/roboticslab-uc3m/developer-manual); hence, [project-generator](https://github.com/roboticslab-uc3m/project-generator) as an implementation that points to it.
1. Take into account https://github.com/roboticslab-uc3m/project-generator/issues
   - We should take https://github.com/roboticslab-uc3m/project-generator/issues/18 (Coexistence with ROS folder structure) into account.
   - We would be answering to https://github.com/roboticslab-uc3m/project-generator/issues/17

PeterBowman, 2018-02-20:
Also: `doc` -> `docs`? I don't want to get too far...

jgvictores, 2018-02-20:
Not sure... I've always read `doc` as `documentation` rather than `documents`.
Perhaps laziness, as there is a considerable amount of links that would have to be updated across organizations.

PeterBowman, 2018-02-20:
Not sure about what feels better, see https://readthedocs.org/. I'd succumb to laziness for this one, though.

> there is a considerable amount of links that would have to be updated across organizations.

I was about to prepare a TODO list, but there are so many repos with `cmake/template/`... +1 for a `tedious` label.

jgvictores, 2018-02-20:
> +1 for a `tedious` label.

Totally agreed: [Ref](https://github.com/asrob-uc3m/actas/issues?q=is%3Aissue+is%3Aopen+label%3A%22Tarea+tediosa+que+nadie+quiere+hacer%22) ((in Spanish, color is meaningful too))

PeterBowman, 2018-03-01:
Reminder: also apply plural rules in Doxyfile! [>>REF<<](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/e1d2a66de87962fc74b17a18fdd6cc123b09956b/doc/Doxyfile#L658)

jgvictores, 2018-03-02:
> Reminder: also apply plural rules in Doxyfile! >>REF<<

Yup!

PeterBowman, 2018-03-15:
The following changes have been applied:

* rename `example/` to `examples/`
* rename `test/` to `tests/`
* rename `cmake/template/` to `cmake/templates/`
* update `doc/Doxyfile`

Affected and **processed** repositories:

* [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics)
* [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices)
* [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins)
* [speech](https://github.com/roboticslab-uc3m/speech)
* [vision](https://github.com/roboticslab-uc3m/vision)
* [tools](https://github.com/roboticslab-uc3m/tools)
* [developer-manual](https://github.com/roboticslab-uc3m/developer-manual)
* [installation-guides](https://github.com/roboticslab-uc3m/installation-guides)
* [teo-main](https://github.com/roboticslab-uc3m/teo-main)
* [teo-developer-manual](https://github.com/roboticslab-uc3m/teo-developer-manual)
* [teo-configuration-files](https://github.com/roboticslab-uc3m/teo-configuration-files)
* [teo-openrave-models](https://github.com/roboticslab-uc3m/teo-openrave-models)
* [asibot-hmi](https://github.com/roboticslab-uc3m/asibot-hmi)
* [asibot-configuration-files](https://github.com/roboticslab-uc3m/asibot-configuration-files)
* [asibot-openrave-models](https://github.com/roboticslab-uc3m/asibot-openrave-models)
* [amor-configuration-files](https://github.com/roboticslab-uc3m/amor-configuration-files)
* [amor-openrave-models](https://github.com/roboticslab-uc3m/amor-openrave-models)
* [tiago-main](https://github.com/roboticslab-uc3m/tiago-main)
* [color-debug](https://github.com/roboticslab-uc3m/color-debug)

Affected and **ignored** repositories:

* [**xgnitive**](https://github.com/roboticslab-uc3m/xgnitive)
* [follow-me](https://github.com/roboticslab-uc3m/waiter)
* [footsteps](https://github.com/roboticslab-uc3m/waiter)
* [force-torque-balance](https://github.com/roboticslab-uc3m/waiter)
* [fpga-nn](https://github.com/roboticslab-uc3m/waiter)
* [fusion-control](https://github.com/roboticslab-uc3m/waiter)
* [gait](https://github.com/roboticslab-uc3m/gait)
* [gaitcontrol](https://github.com/roboticslab-uc3m/gaitcontrol)
* [legacy-matlab-visual-servo](https://github.com/roboticslab-uc3m/waiter)
* [manip-waiter](https://github.com/roboticslab-uc3m/manip-waiter)
* [teo-bimanipulation](https://github.com/roboticslab-uc3m/waiter)
* [teo-grasp](https://github.com/roboticslab-uc3m/waiter)
* [teo-self-presentation](https://github.com/roboticslab-uc3m/waiter)
* [waiter](https://github.com/roboticslab-uc3m/waiter)

No more occurrences found.

jgvictores, 2018-03-15:
THANKS a lot for all of this!

closed 2018-03-16 by PeterBowman

PeterBowman, 2018-06-02:
> Also: doc -> docs? I don't want to get too far...

> Not sure... I've always read doc as documentation rather than documents.
> Perhaps laziness, as there is a considerable amount of links that would have to be updated across organizations.

Just noticed that YCM has a *docs* directory ([ref](https://github.com/robotology/ycm/tree/b6f0566499cbf101b9cc3520b5ab7538b0ad10d4/docs)).