roboticslab-uc3m/questions-and-answers #48: Cache dependencies (e.g. YARP) in Travis

PeterBowman, 2018-02-17:
Thanks to #17, weekly cron jobs automatically test latest commits on production against YARP's `devel` branch. Currently, YARP needs to be cloned and built on every push action, and so do other dependencies (e.g. KDL, PCL, OpenRAVE).

Proposal: for non-cron jobs (perhaps cron too if it makes sense), store YARP (and other dependencies if present) in the Travis cache ([docs](https://docs.travis-ci.com/user/caching/)) in order to speed up things. It's preferrable to hold a global variable with the targeted version so that it can be easily bumped in the YAML file with every new release. Speaking of YARP, current `master` is tagged at 2.3.72.

Caching has been extensively tested on https://github.com/asrob-uc3m/robotDevastation, see [.travis.yml](https://github.com/asrob-uc3m/robotDevastation/blob/10628cc93be3540d9dede26888a95e3666274c49/.travis.yml) and [latest builds](https://travis-ci.org/asrob-uc3m/robotDevastation/builds).

PeterBowman, 2018-02-17:
Another example of caching (CMake + YARP): [roboticslab-uc3m/project-generator](https://github.com/roboticslab-uc3m/project-generator/blob/4e5509d33f9c14fb9088ca4c04eb7c460b5013d3/.travis.yml).

BTW those larger blocks of YAML code ([example](https://github.com/asrob-uc3m/robotDevastation/blob/10628cc93be3540d9dede26888a95e3666274c49/.travis.yml#L68-L91)) could be better placed as standalone shell scripts in `scripts/travis/`.

PeterBowman, 2018-03-20:
Build times have been cut by a half thanks to https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/9bc689c...66032a6 (cache YARP + KDL deps). Two issues arise:

* Since local installations are now preferred for YARP and KDL (in order to store binaries in a cache directory), this procedure has been extended to other deps and the targeted project itself. Now, `LD_LIBRARY_PATH` and `YARP_DATA_DIRS` must be correctly set so that YARP's module loader finds all necessary devices on runtime.
* Native UNIX `dlopen` function does not cope well with this setup. I had to compile YARP with ACE in order to solve that, see https://github.com/roboticslab-uc3m/installation-guides/issues/13#issuecomment-374569031.
* Less `sudo`s led me to think about container-based build on Travis. Sadly, `libace-dev` has not been whitelisted (yet?), see https://github.com/roboticslab-uc3m/installation-guides/issues/13#issuecomment-374571984.

PeterBowman, 2018-03-23:
I'd like to solve https://github.com/roboticslab-uc3m/installation-guides/issues/13 first, but not marking as blocked until further investigation.

As spoken with @jgvictores, let's add another shell script for use by Travis that would extract/deduce the time of creation of the latest cache, then invalidate an regenerate it if older than, say, one week. A new cron job launched once per week would perform this sequence again so that we don't need to wait for the next push build (which may or may not happen in a long time). Also, let's clone and store latest YARP master in this cache instead of a specific release.

PeterBowman, 2018-04-02:
Regarding dependencies, caching and containerization: https://github.com/robotology/yarp/issues/1625.

PeterBowman, 2018-04-06:
> I had to compile YARP with ACE in order to solve that

Fixed at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/ebcfb55df254d68ebbd3df28bbbdd7952d845a76. 

> Less sudos led me to think about container-based build on Travis. Sadly, libace-dev has not been whitelisted (yet?)

No more ACE, and the only relevant `sudo` left in the whole YAML file is [this one](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/ebfe41e7bb016d441b61223050f48b4c021beaaf/.travis.yml#L72). See #50.

PeterBowman, 2018-06-23:
> let's add another shell script for use by Travis that would extract/deduce the time of creation of the latest cache, then invalidate an regenerate it if older than, say, one week

Alternatively, check the latest remote commit against a cached copy, then only invalidate and regenerate the cache if not the last one. Would need a clone operation (other option involves a SSH connection per [this SO answer](https://stackoverflow.com/a/1178586)), but it could improve times significantly when working with non-tagged dependencies such as OpenRAVE.

PeterBowman, 2018-06-29:
Notes:

* `matrix.include` is, apparently, equivalent to `jobs.include` ([ref](https://blog.travis-ci.com/2017-09-12-build-stages-order-and-conditions))
* in dual-compiler builds (gcc and clang), conditional jobs need the compiler name to be explicitly stated, that is:
  ```yaml
  matrix:
    include:
      - if: type = cron
        compiler: gcc
        env: ROBOTOLOGY_CHECKOUT=devel
      - if: type = cron
        compiler: clang
        env: ROBOTOLOGY_CHECKOUT=devel
  ```
  instead of (will only run one job, assuming there is a `compiler:` line somewhere else):
  ```yaml
  matrix:
    include:
      - if: type = cron
        env: ROBOTOLOGY_CHECKOUT=devel
  ```
* conditions and `matrix.exclude` don't cope well together

PeterBowman, 2018-06-29:
> Conditions and matrix.exclude don't cope well together.

Per https://blog.travis-ci.com/2017-09-12-build-stages-order-and-conditions:

> Jobs need to be listed explicitely, i.e. using `jobs.include` or `matrix.include` (alias), in order to specify conditions for them. Jobs created via matrix expansion currently cannot have conditions.

PeterBowman, 2018-06-29:
I have refactored the caching script (now it's a single .sh file that accepts several options via CLI) and applied it to all kin-dyn's dependencies at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/fe405d3c0682a01087a93f220f87eb6e24edfc60.

Notes:

* Still WIP and Travis builds are failing ([ref](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/398464504)).

* Cron jobs will test YARP's devel branch **additionally** to what usual push/pr/api jobs do. Since YARP's master and release jobs are already covered, I'd rather omit them on cron, but it's not that simple (see previous comments).

* Said script uses `wget` to retrieve archived releases, e.g. YARP's v2.3.70.2 tag. I don't know how to make it work with the SSH protocol. This is currently used only for interaction with the amor-api repository, but we are fine as long as we pull latest `develop` (perhaps `master` in the future AMOR v1.0.0 API) only.

PeterBowman, 2018-07-01:
> Since YARP's master and release jobs are already covered, I'd rather omit them on cron, but it's not that simple (see previous comments).

In fact, YARP's master branch builds would make sense once per week, so keep them. Perhaps I could configure Travis in such a way that release builds (e.g. YARP 2.3.70.2, 2.3.72.1, etc.) are added with `matrix.include` and `if: type != cron`, thus they wouldn't be considered for current weekly cron builds.

PeterBowman, 2018-07-08:
Idea: move Travis-related bash scripts (in `scripts/travis/`) to a separate repo (perhaps even a gist?) and clone/download it on every build.

PeterBowman, 2018-07-09:
Remark: [ccache](https://docs.travis-ci.com/user/caching/#ccache-cache) does not seem to improve build times. In fact, they are significantly worse on the first (*cold*) run, and perhaps a bit longer than usual on successive (*warm*) runs.

PeterBowman, 2018-07-09:
Some updates on this:

* Builds have been fixed and caching works like a charm: mean build times have been reduced from ca. 7m40s to ~2m, overall times from ~15m to ~4m, and aggregated times from ~45m to ~13m: [no cache](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/401530985), [with cache](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/401557573).

* Each job generates and restores its own cache storage, which means that the job `YARP_CHECKOUT=master`+gcc does not "see" the cache of `YARP_CHECKOUT=v2.3.70.2`+clang, for instance.

* The previous point means that common caches (e.g. latest KDL-master builds on gcc) are not recicled across similar jobs. Instead, respective copies will be stored in the global (per branch) cache and regenerated several times per job on any upstream change.

* Cache sizes will eventually grow larger with every dropped YARP release. That is, we support YARP 2.3.70.2 today, and will enhance the build matrix with new YARP 3.x releases. However, after removing the YARP 2.3.70.2 builds in the future, the corresponding cache directories will probably persist. Such cases should be handled manually by invalidating/deleting the whole cache.

PeterBowman, 2018-07-09:
> Idea: move Travis-related bash scripts (in scripts/travis/) to a separate repo (perhaps even a gist?) and clone/download it on every build.

Alternatively, use git submodules that point to a branch (usually, they point at a specific commit): https://www.activestate.com/blog/2014/05/getting-git-submodule-track-branch.

PeterBowman, 2018-07-10:
In recent versions of Git, specific branches can be tracked - there is a convenient `branch` property in `.gitmodules` for such purpose. With the new `--remote` option to `git submodule update`, Git will fetch the latest HEAD commit of said branch and merge it into the current submodule tree (see [docs](https://git-scm.com/docs/git-submodule#git-submodule-update--init--remote-N--no-fetch--no-recommend-shallow-f--force--checkout--rebase--merge--referenceltrepositorygt--depthltdepthgt--recursive--jobsltngt--ltpathgt82308203) regarding the `update` command). Still, the hosting repository will track a specific commit instead, which is what I wanted to avoid or circumvent. My original intention was to decouple somehow the revision history of this internal repository and let Travis fetch whatever commit was last at that time. It doesn't make much sense on a second thought, though.

On the other hand, gists can be either cloned (full history) or downloaded (specific or [latest commit](https://gist.github.com/atenni/5604615)). However, they must live in a user account.

PeterBowman, 2018-07-12:
Yes, I just moved the contents of `scripts/travis/` in there. The tests proved that Travis will not update the HEAD commit the submodule points to unless changed manually, which I hoped to avoid by setting a tracking branch in the `.gitmodules` file.

Let's speak about this f2f tomorrow.

PeterBowman, 2018-07-13:
As spoken with @jgvictores, and as a better alternative to using Gists, we could move those scripts to a separate repo (just as I did) and treat it as another hard dependency. Then, Travis would process it in the `before_install` section (either git clone or wget a branch).

Since this is not a priority, (harmless) duplicates of said scripts will be created for now.

PeterBowman, 2018-07-13:
See implementation at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/c597f7a581e34234f0072312a2768513c8915777.

PeterBowman, 2018-07-14:
For future reference (on using `set -e` in shell scripts sourced by Travis): https://github.com/roboticslab-uc3m/tools/commit/2f0fddb96202a20feadc9546e8916e86de9dea34.

PeterBowman, 2018-07-14:
@jgvictores would you like to have this enabled on [xgnitive](https://github.com/roboticslab-uc3m/xgnitive), too? (30 minutes per job ATOW)

jgvictores, 2018-07-14:
That would be very very nice!

PeterBowman, 2018-07-15:
> @jgvictores would you like to have this enabled on xgnitive, too? (30 minutes per job ATOW)

Done at https://github.com/roboticslab-uc3m/xgnitive/commit/8c2f26c9724e16a3b159816c583582f6b6a17ed5. BTW the `Find<pkg>.cmake` scripts were mostly broken and I had to update the installation guides, too: https://github.com/roboticslab-uc3m/installation-guides/commit/1fa60070041afefb183db710a634be2f0bbfe59a. I'm applying the same policy as everywhere else (testing against YARP 2.3.70/2.3.72/master + YARP devel on weekly cron jobs), please adjust it to your needs.

PeterBowman, 2018-07-15:
Done on all Travis-enabled repos except [gait](https://github.com/roboticslab-uc3m/gait). Final notes:

* Codacy proved me wrong in my shell scripting skills ([ref](https://app.codacy.com/app/PeterBowman/robotDevastation/commit?cid=222870784&bid=7117054&utm_campaign=new_commit&utm_medium=Email&utm_source=Internal&), https://github.com/roboticslab-uc3m/questions-and-answers/issues/68), another look at the source code at `scripts/travis/` will do good.
* Certain scripts require a per-repo customization, for instance the *configure-ssh.sh* one (deploy keys).
* Sometimes, spurious cache files for non-default branches will be generated. Another cron-enabled script at `scripts/admin/` (similarly to the doxygen one) might invalidate the cache for said branches on a periodic manner.
* As mentioned in earlier comments, if support is dropped for previous dependency releases, their associated cache directory will remain on server disks. Therefore, the entire cache should be invalidated (e.g. we no longer want to perform tests on YARP 2.3.70, so we remove that line from `.travis.yml`). I wonder if Travis itself invalidates the storage from time to time. Otherwise, another cron script could take care of this (see previous point).

closed 2018-07-15 by PeterBowman

PeterBowman, 2019-04-20:
> As mentioned in earlier comments, if support is dropped for previous dependency releases, their associated cache directory will remain on server disks. Therefore, the entire cache should be [...]

The cache is invalidated after 28 days ([ref](https://docs.travis-ci.com/user/caching/#caches-expiration)):

> Cache archives are currently set to expire after 28 days for open source projects and 45 days for private projects. This means a specific cache archive will be deleted if it wasn’t changed after its expiration delay.

Anyway, that policy is enforced for dangling/old branches, I'm not sure whether the cache files for an active branch (say, develop) would be deleted if not changed - even if used regularly (cron jobs). Manual invalidation is still recommended in the aforementioned case, see https://github.com/roboticslab-uc3m/questions-and-answers/issues/83.