roboticslab-uc3m/questions-and-answers #50: SImplify code coverage in CI builds

PeterBowman, 2018-02-26:
Our Travis lines for code coverage (via Coveralls) look like this ([source](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/3169aa927eae1ea75dbd6058d5619ebc837ee13e/.travis.yml)):

```yaml
install:
  # see http://gronlier.fr/blog/2015/01/adding-code-coverage-to-your-c-project/
  - if [ "$CXX" = "g++" ]; then wget http://ftp.de.debian.org/debian/pool/main/l/lcov/lcov_1.11.orig.tar.gz; fi
  - if [ "$CXX" = "g++" ]; then tar xf lcov_1.11.orig.tar.gz; fi
  - if [ "$CXX" = "g++" ]; then sudo make -C lcov-1.11/ install; fi
  - if [ "$CXX" = "g++" ]; then gem install coveralls-lcov; fi

after_success:
  # capture coverage info
  - if [ "$CXX" = "g++" ]; then lcov --directory . --capture --output-file coverage.info; fi
  # filter out system and test code
  - if [ "$CXX" = "g++" ]; then lcov --remove coverage.info  '/usr/*' 'tests/*' --output-file coverage.info; fi
  # debug before upload
  - if [ "$CXX" = "g++" ]; then lcov --list coverage.info; fi
  # uploads to coveralls
  - if [ "$CXX" = "g++" ]; then coveralls-lcov --source-encoding=ISO-8859-1 coverage.info; fi
```

Perhaps a bit too convoluted if we take a look at official docs:

* https://docs.travis-ci.com/user/coveralls/
* https://github.com/eddyxu/cpp-coveralls#linux (Travis example)

Copy-paste from the latter:

```yaml
language: cpp
compiler:
  - gcc
before_install:
  - pip install --user cpp-coveralls
script:
  - ./configure --enable-gcov && make && make check
after_success:
  - coveralls --exclude lib --exclude tests --gcov-options '\-lp'
```

More links (compiler flags et al.):

* https://stackoverflow.com/questions/16682606/how-to-resovle-gcov-init-undefined-reference-issue-when-link
* https://stackoverflow.com/questions/11783932/how-to-add-linker-or-compile-flag-in-cmake-file
* https://codeflu.blog/2014/12/26/using-gcov-and-lcov-to-generate-beautiful-c-code-coverage-statistics/

jgvictores, 2018-02-27:
Yes. That looks cleaner, clearer, and much more efficient.

PeterBowman, 2018-06-17:
BTW:

```cmake
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fprofile-arcs -ftest-coverage")
```

This also sets linker flags, while the following doesn't (and makes lcov/gcov crash with undefined reference errors, see [this Travis job](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/393358374)):

```cmake
add_compile_options(-fprofile-arcs -ftest-coverage)
```

PeterBowman, 2020-09-01:
Perhaps not worth the trouble if we switch to Codecov at some point: https://github.com/roboticslab-uc3m/questions-and-answers/issues/71.

jgvictores, 2020-09-07:
Not too sure, form a quick glance, Codecov looks different but not automatic: https://github.com/codecov/example-cpp11-cmake/blob/9c09479b7edfdc5e75f29a00e36a94394e688bdf/.travis.yml

PeterBowman, 2021-03-10:
As part of our ongoing transition to GitHub Actions at https://github.com/roboticslab-uc3m/questions-and-answers/issues/91, we decided to drop support for automated code coverage since we had not been using this tool anyway. In case it is deemed convenient to reinstate such support, see modern alternatives and complementary tools to Coveralls such as https://github.com/roboticslab-uc3m/questions-and-answers/issues/71 and https://github.com/roboticslab-uc3m/questions-and-answers/issues/68.

Marking as wontfix and closing due to obsolescence.

closed 2021-03-10 by PeterBowman