roboticslab-uc3m/questions-and-answers #51: Think about AppVeyor (Windows + Linux CI)

PeterBowman, 2018-03-06:
AppVeyor goes Linux according to https://www.appveyor.com/blog/2018/03/06/appveyor-for-linux/.

See how it works in [asrob-uc3m/robotDevastation](https://github.com/asrob-uc3m/robotDevastation):

* [YAML file](https://github.com/asrob-uc3m/robotDevastation/blob/57f65643dcf618ffaba81d6299b89dde698e40e0/.appveyor.yml)
* https://ci.appveyor.com/project/asrob-uc3m/robotdevastation

jgvictores, 2018-03-07:
Looks good, but we should measure where we put our efforts, as at least we have Travis CI.

PeterBowman, 2018-05-15:
In case anyone would like to fiddle with this, here is the follow-up from the AppVeyor blog: https://www.appveyor.com/blog/2018/05/15/appveyor-for-linux-is-generally-available/.

PeterBowman, 2018-10-30:
Funnily, Travis also supports Windows builds, now: https://blog.travis-ci.com/2018-10-11-windows-early-release.

PeterBowman, 2018-11-27:
> Funnily, Travis also supports Windows builds, now: https://blog.travis-ci.com/2018-10-11-windows-early-release.

Tested in https://github.com/roboticslab-uc3m/color-debug/pull/18. Honestly, I don't want to think about AppVeyor anymore. In case we ever need to test our code against Windows, let's try Travis first. If there is any compelling reason (e.g. missing dependencies?), switch to AppVeyor, which in my experience is quite tedious to work with (you need to learn the CMD/PowerShell syntax, no parallel jobs).

closed 2018-11-27 by PeterBowman

jgvictores, 2018-11-27:
Totally agreed. Given the situation with Travis supporting Windows and the extra cons you mention, AppVeyor does not seem the way to go right now.