roboticslab-uc3m/questions-and-answers #52: Fix hard-coded paths of scripts/gnome/___.desktop

jgvictores, 2018-03-18:
Org-wide fix hard-coded paths of scripts/gnome/___.desktop

Learn from https://github.com/roboticslab-uc3m/asibot-configuration-files

Refs:

* [cmake/templates/asibot-apps.desktop.in](https://github.com/roboticslab-uc3m/asibot-configuration-files/blob/935efafd5589daa540e22a47f33916d10ff4f2a6/cmake/templates/asibot-apps.desktop.in)
* [CMakeLists.txt](https://github.com/roboticslab-uc3m/asibot-configuration-files/blob/935efafd5589daa540e22a47f33916d10ff4f2a6/CMakeLists.txt#L42-L51) (no install step yet)

Special treatment as always for https://github.com/roboticslab-uc3m/project-generator

PeterBowman, 2018-03-25:
> CMakeLists.txt (no install step yet)

Let's talk about this. Where should these `.desktop` files land upon install? Is `$HOME/Desktop` a good location, and plain `$HOME` as a fallback in case of trouble? The `xdg-user-dir` utility could come in handy here, see:

* https://askubuntu.com/q/194124
* https://wiki.archlinux.org/index.php/XDG_user_directories
* https://packages.ubuntu.com/trusty/xdg-user-dirs
* https://packages.ubuntu.com/xenial/xdg-user-dirs
* https://github.com/lxde/lxqt-build-tools/blob/4d434ca/cmake/find-modules/FindXdgUserDirs.cmake

jgvictores, 2018-03-25:
> Is $HOME/Desktop a good location, and plain $HOME as a fallback in case of trouble (e.g. missing Desktop directory)?

Yes, that would be perfect.

PeterBowman, 2018-04-06:
> (*) Actually, the generated shortcut should be properly handled to work locally, i.e. from build/. I'd leave this as a TODO.

Done at https://github.com/roboticslab-uc3m/asibot-configuration-files/commit/825b9f72274f2172da6dac0d31980187b3da9861.

PeterBowman, 2018-04-15:
As found out at https://github.com/asrob-uc3m/robotDevastation/issues/126, we are having trouble in correctly displaying the shortcut icon whenever the full path points at system-located PNG files. Apart from that, @jgvictores posted the following link, which may render useful: https://developer.gnome.org/integration-guide/stable/desktop-files.html.en. I learned from it that the *standard* way of registering applications in a GNOME-compatible desktop (so that they show up as menu items (Debian) or as search results in the Unity launcher (Ubuntu)) is to place their `.desktop` shortcuts either in `/usr/share/applications` (all users) or `~/.local/share/applications` (single user). Therefore, we'll may have up to three different copies of the same app shortcut:

* in the build tree of the project
* at `~/Desktop`
* in the *Main Menu* of a GNOME desktop environment or showing up as search results in the Unity launcher

All or some of these might be enabled/disabled via CMake options.

PeterBowman, 2018-04-15:
Solely regarding icon files and their standard location(s): https://askubuntu.com/a/435612.

PeterBowman, 2020-04-26:
Moar on standard directory layout: http://www.pathname.com/fhs/pub/fhs-2.3.html#USRLOCALLOCALHIERARCHY.