roboticslab-uc3m/questions-and-answers #56: grabar video demos

jmgarciah, 2018-04-04:
Se necesita grabar videos de todas las demostraciones que tenemos disponibles ahora mismo para TEO
- teoselfpresetation
- teowaiter
- followme

la idea es subirlos todos al drive de la nueva cuenta de TEO. 

si alguien tiene otros video interesantes de sus investigaciones/experimentos como @RaulFdzbis @elisabeth-ms @David-Estevez @olayasturias , les agradeceria que me los pasara también. Ya que finalmente todos los videos los subirá Edu a la plataforma del Roboticslab.

jmgarciah, 2018-04-04:
para la demo de teowaiter, supongo que será necesario que haya mínimo dos personas.

y para la demos de followme creo que también. además propondría grabar dos videos tanto del robot, como lo que ve el robot, y luego montarlo para que quede chulo ( recomendaciones de @rsantos88 ).

estamos abiertos a otras sugerencias.

jmgarciah, 2018-07-16:
@olayasturias , @elisabeth-ms , @David-Estevez  , @RaulFdzbis , @rsantos88 

Todavía seguimos esperando a que enviéis los vídeos que tengáis para dárselos a Edu.

David-Estevez, 2018-07-16:
> la idea es subirlos todos al drive de la nueva cuenta de TEO.

Existe un drive en el que están recopilados esos vídeos (desconozco si es en la nueva cuenta de TEO o si es otra cuenta). Como comenté en la pasada reunión de TEO, mis vídeos llevan ahí subidos desde hace meses.

jgvictores, 2018-07-16:
Lo de @David-Estevez está en [ROBOTICSLAB-show-press](https://drive.google.com/open?id=0B20131aOHhZubkhydWZZa0M4b00).

David-Estevez, 2018-07-16:
Sí, justo ahí. Gracias por el enlace!

RaulFdzbis, 2018-07-16:
El video de painting que tenemos también esta subido al enlace que pasó Juan (carpeta xgnitive)

rsantos88, 2018-11-06:
Retomo esta issue porque al parecer había cierto interés en subir al canal de [RoboticsLabUC3M](https://www.youtube.com/user/RoboticsLabUC3M) los últimos videos de Teo.

Videos a subir y algunas anotaciones:
* [teo-waiter](https://drive.google.com/drive/folders/0B20131aOHhZuTUxnVEdjSWQ5eDg): aquí hay tres videos separados. ¿Sería interesante editarlos y juntarlos en uno solo para subirlo a youtube?
 * [teo-ironing](https://drive.google.com/open?id=1qQvfcU8-FE8pnMTJ76CTE7tXNcdAZmMF): aquí se subiria el video remix.mp4
* [Push-recovery](https://drive.google.com/drive/folders/0B20131aOHhZuZm15WVRqY1lCUU0) ¿interesa unir estos 2 videos en uno?
* [xgnitive](https://drive.google.com/open?id=0B20131aOHhZuWmstRU9RN1BCSms) aquí se subiría cgda-paint-real-sent-IROS2017.mp4

jgvictores, 2019-03-18:
Hay mucho vídeo grabado, ya metidos en el Google Drive [ROBOTICSLAB-show-press](https://drive.google.com/open?id=0B20131aOHhZubkhydWZZa0M4b00).

Cierro issue, si hay especial interés en subir algún vídeo en particular al canal Youtube [RoboticsLabUC3M](https://www.youtube.com/user/RoboticsLabUC3M), por favor abrir issue (1 por vídeo) en [https://github.com/roboticslab-uc3m/web-and-social-media](https://github.com/roboticslab-uc3m/web-and-social-media)!!

closed 2019-03-18 by jgvictores