roboticslab-uc3m/questions-and-answers #57: Think about https://github.com/jvandemo/copy-github-labels-cli

jgvictores, 2018-04-04:
Per [Regarding which label to assign to an issue in GitHub, we are attempting to unify them, with teo-main for inspiration.](https://github.com/roboticslab-uc3m/developer-manual/blob/1bce22c4f68d2bc80924d68dc3938a702c6da7f2/asking-questions.md): Think about https://github.com/jvandemo/copy-github-labels-cli

jgvictores, 2018-12-12:
Actually using https://github.com/destan/github-label-manager as described at https://medium.com/@dtinth/how-to-copy-github-labels-from-one-project-to-another-1857adc73e0f

Small issues:
1. GitHub `Personal Access Tokens` is currently at https://github.com/settings/tokens
2. Permissions I currently set for this token sum `admin:org, repo`, which are a subset of all possible options, but still sound pretty broad.
3. Sometimes it is tricky, the button is disabled. Sometimes fixed by adding and deleting a fake label (rather than the copy mechanism).
4. Not sure if the token takes some time to enter in effect, but definitely works within the same hour.

Nice that you can just open the page locally.

PeterBowman, 2019-04-21:
Should have posted https://github.com/roboticslab-uc3m/questions-and-answers/issues/76#issuecomment-476727093 here:

> See also https://gist.github.com/Isaddo/7efebcb673a0957b9c6f07cd14826ea4 (a JS web-based label editor).

BTW ZenHub might achieve this too per https://www.zenhub.com/product:

> Save time by updating **labels**, assignees, milestones, and Epics, or make pipeline movements all at once.

jgvictores, 2021-05-10:
Not required, we use https://docs.github.com/en/organizations/managing-organization-settings/managing-default-labels-for-repositories-in-your-organization as commented at https://github.com/roboticslab-uc3m/questions-and-answers/issues/76#issuecomment-570897505 (see https://github.com/organizations/roboticslab-uc3m/settings/labels ).

closed 2021-05-10 by jgvictores