roboticslab-uc3m/questions-and-answers #63: Modificaciones en GitBook

jgvictores, 2018-05-14:
Your organization is being upgraded to the new GitBook" (https://www.gitbook.com/@roboticslab-uc3m/migration/status).
Al parecer han cambiado URLs. Iré actualizando y comentando en esta issue.

jgvictores, 2018-05-17:
Tracking at https://github.com/asrob-uc3m/actas/issues/148 too.

jgvictores, 2018-12-12:
https://github.com/GitbookIO/documentation is deprecated and now points to:
- [Help Center](https://help.gitbook.com/) for general knowledge base. Source is [here](https://github.com/GitbookIO/help.gitbook.com).
- [Toolchain Documentation](https://toolchain.gitbook.com/) for complete documentation of GitBook format and tools. Source is in the [official repository](https://github.com/GitbookIO/gitbook/)

jgvictores, 2018-12-12:
- https://github.com/GitbookIO/gitbook/blob/master/docs/plugins/hooks.md
- https://github.com/GitbookIO/gitbook/issues/684

jgvictores, 2018-12-13:
One of the first steps is running Gitbook on localhost: both versions if possible (and also find out if legacy is really legacy or just legacy from the perspective of https://www.gitbook.com).

jgvictores, 2018-12-14:
Turns out that https://github.com/GitbookIO/gitbook is the open source command line tool and Node.js library that generated the beautiful `legacy` books.

From "How will this project move forward - is it left for dead? [GitbookIO/gitbook#2148](https://github.com/GitbookIO/gitbook/issues/2148)" we read that <https://www.gitbook.com> has decided to move away from open source. At time of writing:
- `legacy`: https://asrob-uc3m.gitbooks.io/robotdevastation-user-manual/content/en/
- `new`: https://asrob-uc3m.gitbook.io/robotdevastation-user-manual

|             |source  | arrow keys  | languages |  interface |
|-----------|----------|-----------|-------------|----------|
| `legacy` | open    |   yes      | yes        | clean   |
| `new`     | closed |      no   | no         | crowded |

We've began a a list of alternatives starting [here](https://github.com/GitbookIO/gitbook/issues/2148#issuecomment-434592652). However, I'll continue to use this issue to track progress on `legacy`, which was nice.

PeterBowman, 2018-12-14:
More alternatives to be found in https://alternativeto.net/software/gitbook.

jgvictores, 2018-12-14:
`legacy`, at least for generating `.html`, works great:
1. Install via: npm install gitbook-cli -g
2. Then two options:
   1. Generate working `_book/` via `gitbook build`
   2. Serve via: `gitbook serve` (generates `_book` anyway)

Small note: On a PC without `apache`, had to use the `gitbook serve` option: simply opening the `index.html` rendered but arrow keys and links did not work. I've been running into similar errors lately, where very plain client-side code had to be served... Turns out I may have been running into "For security purposes, Mozilla applications block links to local files (and directories) from remote files. This includes linking to files on your hard drive, on mapped network drives, and accessible via Uniform Naming Convention (UNC) paths. " as described at [S.O.](https://stackoverflow.com/questions/32563078/how-link-to-any-local-file-with-markdown-syntax) (also for Chromium). Tried serving the generated `_folder` via `apache` on server, everything cool. Simple and fast.

jgvictores, 2018-12-15:
https://help.github.com/articles/using-a-static-site-generator-other-than-jekyll/ says GitHub pages will host, but not generate, alternatives to Jekyll. We'd have to rely on a server (think about heroku).

jgvictores, 2018-12-17:
Turns out that `legacy` gitbook.com URLs for `robotics-lab` are dead due to the update (we had been warned there was no way back). Example of broken link as analogous to [here](https://github.com/roboticslab-uc3m/questions-and-answers/issues/63#issuecomment-447504747) (we've tried several of these): https://roboticslab-uc3m.gitbooks.io

jgvictores, 2018-12-20:
No longer blocking, provided these results.

jgvictores, 2019-01-02:
New `README-gitbook.md` with banner at:
- https://github.com/roboticslab-uc3m/developer-manual/commit/37e9d3900dc1a11e82547e369f52f70f93bc59af
- https://github.com/roboticslab-uc3m/teo-developer-manual/commit/b4498cc1129874df377c654fff194d8e5200dd80

jgvictores, 2019-01-02:
> - [x] set up temporal cron jobs enabling temporal URLs at robots.uc3m.es

- https://github.com/roboticslab-uc3m/developer-manual/commit/c5d6a439858f376f43282c1ba84f99d0e081335d
- https://github.com/roboticslab-uc3m/teo-developer-manual/commit/c91ea9ed1a45a62b880d4b94ec13a89f623dbd6a

jgvictores, 2019-01-03:
Pending: installation-guides

jgvictores, 2019-01-11:
> Pending: installation-guides

See https://github.com/roboticslab-uc3m/installation-guides/issues/58

PeterBowman, 2021-03-12:
For future reference, this is installed on robots.uc3m.es at time of writing:

```bash
$ gitbook --version
CLI version: 2.3.2
GitBook version: 3.2.3
```

jgvictores, 2021-03-17:
> ```shell
> $ gitbook --version
> CLI version: 2.3.2
> GitBook version: 3.2.3
> ```

Which, as commented at http://robots.uc3m.es/gitbook-installation-guides/install-gitbook.html ([perma](https://github.com/roboticslab-uc3m/installation-guides/blob/6f5c71dc5ad4852f2a27ee8cc43a3c93c21f3d95/install-gitbook.md)) works with `nodejs 10.x`.