roboticslab-uc3m/questions-and-answers #72: Upcoming changes in Linux build infrastructure on Travis CI

PeterBowman, 2018-10-30:
We moved on to container-based (`sudo: false`) builds on Travis as a result of https://github.com/roboticslab-uc3m/questions-and-answers/issues/48. The other alternative are virtualized builds (`sudo: required`). There is an ongoing transition process to migrate projects from the former to the latter. Soon, all `sudo: false` configs will be disregarded and treated as `sudo: required`.

https://blog.travis-ci.com/2018-10-04-combining-linux-infrastructures

> Starting in the second week of November (we’ll post another blog post), we will start to migrate repositories that use `sudo: false` to the virtual machine-based infrastructure.

jgvictores, 2018-10-30:
Not much to do here, right?

PeterBowman, 2018-10-30:
Just make sure that we may follow this transition smoothly and update our .travis.yml everywhere (hence the "tracking" label).

PeterBowman, 2018-10-30:
Despite being close to the final term (mid-November), the information is too scarce for now, so I'd wait for the next blog post they promised. Docs are still to be updated, but there is a short note at https://docs.travis-ci.com/user/reference/overview/#for-a-particular-travisyml-configuration:

> Between middle of October 2018 and end December 2018 the default infrastructure your builds runs on will depend on a few different factors while we consolidate everything onto sudo-enabled virtual machines.

PeterBowman, 2018-11-27:
https://blog.travis-ci.com/2018-11-19-required-linux-infrastructure-migration

> In the next phase of the migration, all builds will run on virtual-machine-based infrastructure – regardless of the configuration for `sudo` in the `.travis.yml`. If you currently specify `sudo: false` in your `.travis.yml`, we recommend removing that configuration soon.

> The timeline for this migration will be as follows:
>
> * 19 November, 2018 - Today we publish this post and are ready to answer all your questions!
> * 28 November, 2018 - We will send a service email to remind folks still using `sudo: false` on recent builds to remind you to migrate.
> * 03 December, 2018 - We will start randomly sampling projects on both travis-ci.org and travis-ci.com to move them permanently to using the virtual-machine-based infrastructure for all builds. The projects will be migrated incrementally over a few days
> * 07 December, 2018 - All projects that use a Linux build environment will be fully migrated to using the same Linux infrastructure, which runs builds in virtual-machines.

closed 2018-12-03 by PeterBowman