roboticslab-uc3m/questions-and-answers #76: Define standard for issue labels

jgvictores, 2018-12-18:
Define standard for issue labels.

Additionally specify if any specific repository/project will have a different set of labels.

Personal thoughts which lead to no specific direction:
- At #57 we found out https://github.com/destan/github-label-manager is not perfect, but useful for copying issues
- I've been doing some priority (idea/low/high/critical) and complexity (easy/hard/tedious) labeling myself, which I think is related to scrum/agile but the main motivation is it really helps sort things out.
- At https://github.com/roboticslab-uc3m/developer-manual/blob/master/asking-questions.md ([perma](https://github.com/roboticslab-uc3m/developer-manual/blob/51d2cb68589a23fb22df86d3682ac47d1c1a00b5/asking-questions.md)) `teo-main` establishes the reference pattern. We can speak about this!
   - Maybe https://github.com/roboticslab-uc3m/project-generator/labels is cooler!

jgvictores, 2019-01-08:
Some things commented with @David-Estevez (very related to #74)

1. We use spaces. Example: `priority: critical`
2. Remove `status: in progress` as soon as we are sure we can access kanban column info via GitHub API.
3. Under same premise as (2), remove `blocked`, rename `blocking`->`priority: blocking`
4. `regression: upstream` is simply a `regression`, we are the ones who have to fix it anyway.
5. For labels, we define the basics, but new can be added per repo.

jgvictores, 2019-01-09:
Additional note: For handling blocked/blocking issues, it seems like the ideal solution would neither be labels nor project columns (see #74), but instead a dependency tree. :smile:

PeterBowman, 2019-01-19:
+1 for a `regression` label.

jgvictores, 2019-02-08:
Notes:
- https://github.com/roboticslab-uc3m/kinematics-dynamics/labels/super%20cool
- https://github.com/roboticslab-uc3m/questions-and-answers/labels/Pending%20%3Asmile%3A2%3Asmile%3A%20meeting

jgvictores, 2019-02-09:
Lots already at https://github.com/roboticslab-uc3m/developer-manual/blob/master/best-practices-github.md#github-issues (rendered at http://robots.uc3m.es/gitbook-developer-manual/best-practices-github.html#opening-a-github-issue)

PeterBowman, 2019-03-19:
> I was thinking of splitting `upstream` to describe two situations:

Also: broken code due to an upstream bug, found local workaround, local (RL-side) issue closed (https://github.com/roboticslab-uc3m/WBPCwaiter/issues/2#issuecomment-474417097).

PeterBowman, 2019-03-26:
See also https://gist.github.com/Isaddo/7efebcb673a0957b9c6f07cd14826ea4 (a JS web-based label editor).

PeterBowman, 2019-03-27:
Idea: rename *tracking* (an "issue of issues") to *horizontal* (affects multiple repos, perhaps even orgs).

jgvictores, 2019-04-30:
> Idea: rename _tracking_ (an "issue of issues") to _horizontal_ (affects multiple repos, perhaps even orgs).

I'll begin to apply this

PeterBowman, 2019-12-28:
See also: org-wide labels!

https://help.github.com/en/github/setting-up-and-managing-organizations-and-teams/managing-default-labels-for-repositories-in-your-organization

jgvictores, 2020-01-05:
> https://help.github.com/en/github/setting-up-and-managing-organizations-and-teams/managing-default-labels-for-repositories-in-your-organization

Pretty cool. Looks like it only applies on creation of a new repo, but still very useful.

Adding [teo-main/labels](https://github.com/roboticslab-uc3m/teo-main/labels) to https://github.com/organizations/roboticslab-uc3m/settings/labels

jgvictores, 2020-01-13:
Similar issue on `robotology` org: https://github.com/robotology/robotology.github.io/issues/10

jgvictores, 2021-05-10:
> Adding [teo-main/labels](https://github.com/roboticslab-uc3m/teo-main/labels) to https://github.com/organizations/roboticslab-uc3m/settings/labels

Closing via :top:

closed 2021-05-10 by jgvictores