roboticslab-uc3m/questions-and-answers #77: Migrate alternative build systems to cmake

jgvictores, 2019-01-18:
Migrate alternative build systems to cmake, essentially scripts, such as:
- doxygen
- gitbook

PeterBowman, 2019-04-19:
>     * doxygen

Done? See https://github.com/roboticslab-uc3m/questions-and-answers/issues/79.

PeterBowman, 2019-04-23:
Yes, Doxygen is ready. we could abstract the generation of gitbooks (see example [update-gitbook.sh](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/13d36106fd6a9c45e986e7ba203520ee8c71151a/scripts/admin/update-gitbook.sh) script) with CMake, but we lack a proper CMake wrapper (e.g. there is a FindDoxygen, but no FindNode nor FindGitbook). Since cooking our own solution feels like overkill, let's not meddle with this any longer.

Conclusion: would be nice to have a `make dox`, but we are happy with a simple shell script (for now!).

closed 2019-04-23 by PeterBowman