roboticslab-uc3m/questions-and-answers #78: YCM 0.10 upgrade path

PeterBowman, 2019-02-28:
Currently in devel branch ([release notes](https://github.com/robotology/ycm/blob/1a231f296f859b4e476707fd655e5c5df7c98ce4/help/release/0.10.0.rst)):

- [ ] The [FetchContent module](https://cmake.org/cmake/help/v3.11/module/FetchContent.html) from CMake 3.11 will be imported via cmake-next. We might want to fetch googletest on configure time (already moved from embedded copy to apt package or on-system installation), see https://github.com/roboticslab-uc3m/questions-and-answers/issues/41#issuecomment-467911707.

- [ ] The [ExternalProject module](https://cmake.org/cmake/help/v3.14/module/ExternalProject.html) from CMake master was locally updated into cmake-next/proposed, allegedly most customizations had been removed. We use the `CONFIGURE_SOURCE_DIR` option, which does not exist anymore, in the [Buildorocos_kdl.cmake script](https://github.com/roboticslab-uc3m/asibot-main/blob/0259f65fb5c7043319ca0f8d0cc212f15d171b09/cmake/ycm-modules/Buildorocos_kdl.cmake#L7) across superbuild projects. I suspect this may break once YCM 0.10 is merged into master.

PeterBowman, 2019-03-04:
>     * I suspect this may break once YCM 0.10 is merged into master. The `CONFIGURE_SOURCE_DIR` might need to be replaced with `SOURCE_DIR`, not sure though.

It does break, indeed. The upstream CMake option should read `SOURCE_SUBDIR`. Reported at https://github.com/robotology/ycm/issues/210.

PeterBowman, 2019-03-06:
> It does break, indeed.

My bad, got fooled by the way CMake treats local and cached variables. See https://github.com/robotology/ycm/issues/210#issuecomment-470091529 for context.

PeterBowman, 2019-03-07:
> The ExternalProject module from CMake master was locally updated into cmake-next/proposed [...]

TODO: replace deprecated option (needs YCM 0.10) and apply workaround given at https://github.com/robotology/ycm/issues/212 (can/should be done now).

PeterBowman, 2019-03-18:
> The [ExternalProject module](https://cmake.org/cmake/help/v3.14/module/ExternalProject.html) from CMake master was locally updated into cmake-next/proposed, allegedly most customizations had been removed. We use the `CONFIGURE_SOURCE_DIR` option

> TODO: replace deprecated option (needs YCM 0.10) and apply workaround given at [robotology/ycm#212](https://github.com/robotology/ycm/issues/212) (can/should? be done now).

No hurry, our current code will work fine thanks to the upstream workaround. This bullet boils down to replacing the will-be-deprecated option to `ycm_ep_helper()`.

PeterBowman, 2019-04-01:
YCM v0.10 is out! https://github.com/robotology/ycm/releases/tag/v0.10.0

PeterBowman, 2019-04-08:
YCM v0.10.1 was released today. We might have found another issue: https://github.com/robotology/ycm/issues/244.

PeterBowman, 2019-04-18:
> We might have found another issue: [robotology/ycm#244](https://github.com/robotology/ycm/issues/244).

Fixed in current YCM master branch, scheduled for v0.10.2. This has fixed our superbuild Travis jobs, too.

PeterBowman, 2019-04-19:
> The [FetchContent module](https://cmake.org/cmake/help/v3.11/module/FetchContent.html) from CMake 3.11 will be imported via cmake-next. We might want to fetch googletest on configure time (already moved from embedded copy to apt package or on-system installation), see [#41 (comment)](https://github.com/roboticslab-uc3m/questions-and-answers/issues/41#issuecomment-467911707).

I can't think of a good solution, see https://github.com/roboticslab-uc3m/questions-and-answers/issues/41#issuecomment-484919930. The FetchContent way is kinda close to YCM superbuilds, which I wanted to avoid. On the other hand, it gives total control of the googletest version requested, but we'd like to look for it on system first (the YCM-way) and fall back to the download step - this option involves a custom FindGTestSources script, again.

Nothing more to do here, closing.

closed 2019-04-19 by PeterBowman

PeterBowman, 2019-10-23:
> Drop install-related lines in superbuild-repos ([ref](https://github.com/roboticslab-uc3m/asibot-main/blob/0259f65fb5c7043319ca0f8d0cc212f15d171b09/CMakeLists.txt#L35-L40)): [robotology/ycm#235](https://github.com/robotology/ycm/pull/235) (experimental?).

Will be made relocatable, see https://github.com/robotology/ycm/pull/289.

PeterBowman, 2023-12-23:
The `GIT_SHALLOW` and `GIT_PROGRESS` were actually ignored since `ycm_ep_helper` never forwarded them to `ExternalProject_Add`. Support has been recently added for a `SHALLOW` option (https://github.com/robotology/ycm/pull/440), I'm going to use this instead.

traversaro, 2023-12-23:
@PeterBowman if you need more arguments to be forwarded to ExternalProject feel free to suggest them!