roboticslab-uc3m/questions-and-answers #81: Embrace hooks

PeterBowman, 2019-04-01:
According to [Wikipedia](https://en.wikipedia.org/wiki/Hooking), "hooking" is referred to as follows:

> In computer programming, the term hooking covers a range of techniques used to alter or augment the behaviour of an operating system, of applications, or of other software components by intercepting function calls or messages or events passed between software components. Code that handles such intercepted function calls, events or messages is called a hook.

Check also [StackOverflow](https://stackoverflow.com/questions/467557/what-is-meant-by-the-term-hook-in-programming):

> Essentially it's a place in code that allows you to tap in to a module to either provide different behavior or to react when something happens.

In that vein, I deem it's time to adopt "hooks" in our codebase. We can arrange a proper place for developers to hook up and share their ideas. Whenever troubles arise, experienced programmers will easily get off the hook, whereas newcomers are prompted to hook into this matter as soon as possible. By hook or by crook, it's so easy to get hooked on!

[image]

PeterBowman, 2019-04-01:
Recent studies show that Brainfuck has proven to provide a seamless way to implement *hooks*, see https://github.com/robotology/yarp/pull/1988.

jgvictores, 2019-04-01:
Gettin' the (...) of it at https://github.com/roboticslab-uc3m/questions-and-answers/blob/cd6dd748f2262f1e1d2c9cb0b90bad3488782264/hook.stl

PeterBowman, 2019-04-05:
Closing until we get a proper [Parrot](https://www.parrot.com/en/minidrones/parrot-jumping-sumo) to complement this new pirate theme.

closed 2019-04-05 by PeterBowman