roboticslab-uc3m/questions-and-answers #82: Bump required YCM version (v0.10+)

PeterBowman, 2019-04-02:
We are about to increase the required version of YCM as a hard dependency in several repos. On the next few days, YCM v0.10 or later will be requested by root CMakeLists.txt following the progress of https://github.com/roboticslab-uc3m/questions-and-answers/issues/78 and https://github.com/roboticslab-uc3m/questions-and-answers/issues/79. We acknowledge that a bug spoils superbuilds (&lt;robot&gt;-main repositories) if YCM is pulled from the web; a fix is expected and coming to v0.10.1, soon (https://github.com/robotology/QA/issues/338#issuecomment-478552030).

traversaro, 2019-04-02:
You may be interested in this regression as well: https://github.com/robotology/robotology-superbuild/issues/191 .

PeterBowman, 2019-04-02:
@traversaro thanks!

PeterBowman, 2019-04-19:
cc @roboticslab-uc3m

YCM 0.10 or later is now enforced in several projects, specific requirements are detailed in the installation instructions at `doc/<project>-install.md`.

> We acknowledge that a bug spoils superbuilds (&lt;robot&gt;-main repositories) if YCM is pulled from the web; a fix is expected and coming to v0.10.1,

More issues arose in YCM v0.10.1, those are fixed in YCM master branch and a v0.10.2 release is expected, soon. If you are using a &lt;robot&gt;-main repository, make sure YCM performs the update-self step (alternatively, delete the `external/YCM/` directory).

closed 2019-04-19 by PeterBowman

PeterBowman, 2019-04-26:
[YCM 0.10.2](https://github.com/robotology/ycm/releases/tag/v0.10.2) has been released today with several bugfixes, it is recommended to upgrade from 0.10.0 or 0.10.1.