roboticslab-uc3m/questions-and-answers #84: Update CMake/YCM export options

PeterBowman, 2019-08-14:
YCM 0.11 will introduce a bunch of new features regarding how targets are treated upon export:

* https://github.com/robotology/ycm/issues/262: `set(CMAKE_EXPORT_PACKAGE_REGISTRY ON)` instead of calling `export(PACKAGE)` explicitly ([example](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/631f68df6796e132a8ea5ac473f7031b1c3c988d/CMakeLists.txt#L93-L94)).
  * addects most repos
* https://github.com/robotology/ycm/issues/263: add `NO_EXPORT` option to `install_basic_package_files()` in order to instruct YCM not to export targets when there are actually none ([example](https://github.com/roboticslab-uc3m/speech/blob/bf1a83cc13fa3320bb01be4295446aa7065a5957/CMakeLists.txt#L140-L153)). Addresses https://github.com/roboticslab-uc3m/project-generator/issues/19.
  * [speech](https://github.com/roboticslab-uc3m/speech/blob/bf1a83cc13fa3320bb01be4295446aa7065a5957/CMakeLists.txt)
  * [vision](https://github.com/roboticslab-uc3m/vision/blob/9c283dd40807f8f90b70bb31a5b0a9fd77bc88a3/CMakeLists.txt)
  * [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/2801f197b35f60eacdbdfd6214ef67d5db99272a/CMakeLists.txt)
  * [teo-bimanipulation](https://github.com/roboticslab-uc3m/teo-bimanipulation/blob/1a1c928b6cc84cdb2d641e04cd647e44e35d8562/cmake/TEO_BIMANIPULATIONDescribe.cmake)
  * [asibot-hmi](https://github.com/roboticslab-uc3m/asibot-hmi/blob/b1c10eb8f1b85897e786d28505dc14333ae874e7/cmake/ASIBOT_HMIDescribe.cmake)
  * [project-generator](https://github.com/roboticslab-uc3m/project-generator/blob/c8913e04e7896efb164d89ad96cabaa39951f7cd/cmake/TEMPLATE_NAMEDescribe.cmake)
  * [tiago-main](https://github.com/roboticslab-uc3m/tiago-main/blob/af28f4af71d2d8f7f735f8b0e52bc93975547dcb/cmake/TiagoDescribe.cmake)
  * **note:** make sure to update/remove *&lt;NAME&gt;Config.cmake.in*!

PeterBowman, 2019-10-02:
Seems like YCM 0.11 might bump minimum required CMake version to 3.12: https://github.com/robotology/ycm/issues/261. Thus, requiring YCM 0.11+ in our repos would break support for Ubuntu 16.04 unless users are instructed to pull the latest (CMake 3.15) release from the new https://apt.kitware.com/ repository.

PeterBowman, 2019-10-04:
See announcement and a brief CMake 3.12 migration guide at https://github.com/robotology/QA/issues/364.

PeterBowman, 2019-11-28:
YCM 0.11 has been released today: https://github.com/robotology/ycm/releases/tag/v0.11.0.

PeterBowman, 2019-12-06:
Also, YCM 0.11 introduces the CMake option `YCM_EP_INSTALL_DIR` to change where the subprojects will be installed (default: `install/`). I'd consider this on superbuild repos, e.g. [asibot-main](https://github.com/roboticslab-uc3m/asibot-main/blob/657c3562358956da2ebafd127d6c606650d14dfe/.travis.yml#L34).

traversaro, 2019-12-06:
> Also, YCM 0.11 introduces the CMake option `YCM_EP_INSTALL_DIR` to change where the subprojects will be installed (default: `install/`). I'd consider this on superbuild repos, e.g. [...]

Just to let you know, the variable `YCM_EP_INSTALL_DIR` is quite experimental, and we did not test it extensively. In particular, I am not really sure if it works correctly `YCM_EP_INSTALL_DIR` is a prefix that also contains some libraries that are used as dependency of the superbuild.

PeterBowman, 2019-12-06:
@traversaro that's good to know, thanks!

PeterBowman, 2019-12-14:
https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/4b1f60d...ee572ce

PeterBowman, 2020-01-29:
Issues with CMake 3.15+: https://github.com/robotology/ycm/issues/262#issuecomment-579840557.

PeterBowman, 2020-03-16:
Relevant repositories have been updated. Closing.

closed 2020-03-16 by PeterBowman