roboticslab-uc3m/questions-and-answers #87: Changes to git branch policy in YARP/YCM

PeterBowman, 2019-12-06:
Changes to git branch policy in YARP/YCM force us to rethink the way we test our software against robotology components: https://github.com/robotology/QA/issues/372.

Fast recap:

* devel branches will disappear at some point, currently they are fast-forwarding the master branch
* all experimental development leading to the next major/minor release is now merged into master
* new stable branches per minor release, e.g. yarp-3.3 for the whole release train of YARP v3.3.x
* I presume tags will continue being produced per patch release, e.g. tags/v3.3.0

Disclaimer (announcement-like, cc @roboticslab-uc3m): it is encouraged to checkout a stable branch upon git clone of either YARP or YCM, e.g.:
  * `git clone https://github.com/robotology/yarp; cd yarp; git checkout yarp-3.3`, or
  * `git clone --branch=yarp-3.3 https://github.com/robotology/yarp`.

Consequences regarding Travis CI:

* https://github.com/roboticslab-uc3m/questions-and-answers/issues/17: our cron jobs need to address YARP/YCM master branch instead of devel
* https://github.com/roboticslab-uc3m/questions-and-answers/issues/48: the job matrix should now target the new minor release branches, therefore we'll no longer pull older YARP releases from the GitHub archive but directly clone the specific branch instead (mind https://github.com/roboticslab-uc3m/questions-and-answers/issues/86)

On a side note, perhaps we should be slowly moving towards [GitHub Actions](https://github.com/actions), provided this might eventually replace Travis CI (I'm not sure about it).

PeterBowman, 2019-12-06:
Weekly cron jobs on YARP/YCM devel are now disabled.

PeterBowman, 2019-12-06:
BTW all master/yarp-3.3 builds will fail inevitably due to a Clang bug: https://github.com/robotology/yarp/issues/2086. Also, latest OpenRAVE release introduced a regression: https://github.com/roboticslab-uc3m/installation-guides/issues/65#issuecomment-544455403.

jgvictores, 2019-12-07:
> * `git clone --branch=yarp-3.3 https://github.com/robotology/yarp`.

Added at https://github.com/roboticslab-uc3m/installation-guides/commit/ea11b4f088ccf0f7ceeb06e3f364e8d2242ef4dd

> * both YCM and YARP (which consumes YCM) now require CMake 3.12+, see [#84 (comment)](https://github.com/roboticslab-uc3m/questions-and-answers/issues/84#issuecomment-561341909)

https://github.com/roboticslab-uc3m/installation-guides/commit/ea11b4f088ccf0f7ceeb06e3f364e8d2242ef4dd and https://github.com/roboticslab-uc3m/installation-guides/commit/da0a2d13d0c8715c95bc5a695cbf91a259cd740d

PeterBowman, 2019-12-09:
> On a side note, perhaps we should be slowly moving towards [GitHub Actions](https://github.com/actions), provided this might eventually replace Travis CI (I'm not sure about it).

Robotology folks have mastered GH Actions in the [robotology-superbuild repo](https://github.com/robotology/robotology-superbuild), see [.github/workflows/ci.yml](https://github.com/robotology/robotology-superbuild/blob/beaf06287a4f0a0496bd11dc6e1fc2da2172a00e/.github/workflows/ci.yml) and [/actions](https://github.com/robotology/robotology-superbuild/actions). I see full support for builds on Ubuntu+macOS+Windows, which is very nice.

PeterBowman, 2019-12-10:
> BTW

All TEO PC's are now running latest CMake from Kitware's apt repository and YCM v0.11 + YARP v3.3. Also, I have upgraded the PeakCAN driver&lib combo to version 8.9.3 (latest).

PeterBowman, 2019-12-14:
https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/4b1f60d...ee572ce

PeterBowman, 2019-12-19:
Changes:

* all Travis builds request Bionic images, now
* *git-cache-dependency.sh* script was updated
* no longer testing *devel* branches on weekly cron jobs, using *master* instead
* push jobs fetch and cache last two YARP release branches, currently *yarp-3.3* and *yarp-3.2*, and latest YCM release (*ycm-0.11* ATOW)
* superbuild repos run weekly crons on *develop* unless there has been another build in the last 24h

closed 2019-12-19 by PeterBowman

PeterBowman, 2020-09-09:
> Drawbacks: the vision repo fails to compile due to a weird preprocessor error ([build](https://travis-ci.com/roboticslab-uc3m/vision/builds/141118440)).

Fixed at https://github.com/roboticslab-uc3m/vision/commit/37ce63dce91670716cac5c687ae08040214149fd.