roboticslab-uc3m/questions-and-answers #90: Using yarp's internal logging for yarp devices

David-Estevez, 2020-10-10:
Hi!

I have just discovered that yarp has a [logging system for the log messages](http://www.yarp.it/Log_8h_source.html) (info, debug, warning, error...) of their yarp devices. We are using a [similar system](https://github.com/roboticslab-uc3m/color-debug) for our devices, based on `color-debug`, but I guess their logging system has better integration with yarp devices and other yarp mechanisms. Messages would also look more homogeneous if sent to stdout with the yarp logger instead of our own system.

The question is... should we use yarp's internal logging for our devices? or should we stick to the current `color-debug` system?

PeterBowman, 2020-10-11:
<details>
  <summary>Regular output</summary>

  [image]
</details>

<details>
  <summary>Yay, colors</summary>

  [image]
</details>

<details>
  <summary>Compact output (colored markers as log levels)</summary>

  [image]
</details>

<details>
  <summary>Verbose output</summary>

  [image]
</details>

<details>
  <summary>yarplogger GUI</summary>

  [image]
</details>

PeterBowman, 2020-10-11:
Further remarks (see https://github.com/roboticslab-uc3m/questions-and-answers/issues/90#issuecomment-517458703):

* Logging utilities support "components", i.e. additional tags that help identify the source of a log line. These are placed between the log level and the actual log message, e.g. `[INFO] |yarp.os.Port| Port /bla active...`
* Trace lines are dismissed by the compiler unless the project is build in *Debug* configuration. Besides, they need to be explicitly enabled through an environment variable. In non-verbose mode, trace lines also include the calling method's signature.
* For *yarplogger* to work, applications must be launched either via `yarprun` (initialized with the `--log` parameter) or command line plus an environment variable:
  * *yarprun* enables launching apps on remote machines, optionally piping the standard output to an xterm shell. If enabled, any kind of output is forwarded to the logger; this includes YARP logger utilities, `printf`, `std::cout`...
  * For *yarplogger* to collect application output when launched via command line, the `YARP_FORWARD_LOG_ENABLE` variable needs to be set to `1`. This comes at the cost of using another port/thread and so on, and only YARP logger utilities can be piped (no other output is allowed).
* *yarplogger* has plenty of funcionalities, you can even unveil the full backtrace of each line if explicitly enabled, along with the absolute path of the source file, line number, calling method, etc.
* The CLI counterpart, *yarplogger-console*, is mostly an RPC interface for quickly inspecting available log streams and saving logs to file, which is of course also attainable via regular *yarplogger*.
* Stream macros are slightly less performant than the C-style ones, but also easier to use. For instance, there is an overload for printing a `std::vector`.
* We can emulate the old `CD_XXX_NO_HEADER` behavior with stream macros:
  ```c++
  std::vector<int> my_vector { 1, 2, 3, 4, 5 };
  {
      auto _temp = yInfo() << "let's unpack a random vector:";
      for (int value : my_vector)
          _temp << value;
      _temp << "end";
  }
  // logger prints this before leaving the previous code block:
  // let's unpack a random vector: 1 2 3 4 5 end
  ```

PeterBowman, 2020-10-11:
I have given this a try at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/ea405eb126840438b386fcfbadb50174796f3559 and https://github.com/asrob-uc3m/yarp-devices/commit/c7f717c5b8dc025a383f6123f6aeb7133b0b3f6c. Some things to consider:

* Component names should be expressive, but not too long. I have chosen the schema `namespace.main_class`, much like core YARP does. I refused to add an intermediate term for the repository name (in my initial thoughts, the schema was `github_org.github_repo.main_class`) since that would make much longer names. Also, I assume name clashes should not be a thing in our orgs and, if necessary, a device/library/application can either refine its component name a bit or use more than one component.

* Log contents should be inspected one by one to make sure their original intention is preserved. Now, color-debug provides the file name and line number along with the log level and the log message. This filename is not always unique, but we manage to sort things out. At most, and when not in verbose mode, YARP logging utilities can only provide the component name via stdout/stderr. It is therefore a good idea to enhance existing log messages (where deemed convenient) so that they reflect the method or process currently going on.

* In the light of the thorough review I am anticipating in the previous point, it should be a good idea to check the logic of existing log levels (i.e. is this debug line really debug, or should I promote it to info level, and so on) and whether more log lines should be added. There is one particular case: we abuse `CD_DEBUG` for signalizing method entrance. It might be a good idea to switch to `yTrace` here, unless we really need them in Release config. Alternatively, throttling macros could be an option.

* As an additional side quest, search for `std::cout`s and `printf`s that should be morphed into YARP logging macros. Also, make sure there are no newlines in the newly adapted YARP log lines, since those are automatically appended anyway.

PeterBowman, 2020-10-22:
Note the `--verbose` option to `yarpdev` and several other companion utilities has been superseded by logging components (https://github.com/robotology/yarp/issues/2389#issuecomment-714720180). This may partially affect the solutions adopted in https://github.com/roboticslab-uc3m/questions-and-answers/issues/49. For instance, but not sure if a feature or a bug, I see all debug lines related to device configuration when issuing `yarpdev --device EmulatedControlboard --name /fake`, no `YARP_VERBOSE=1` involved (this would bloat the console with lots of comms-related notices).

PeterBowman, 2021-05-10:
Color-debug is officially dead in most of roboticslab-uc3m ([search](https://github.com/search?l=C%2B%2B&q=org%3Aroboticslab-uc3m+colordebug&type=Code)) and in all asrob-uc3m repos ([search](https://github.com/search?p=2&q=org%asrob-uc3m+COLOR_DEBUG&type=Code)). Leftovers:

- https://github.com/roboticslab-uc3m/WBPCwaiter (PhD thesis-related): demo-like repo
- https://github.com/roboticslab-uc3m/teo-blender-models: just in a single C++ file
- https://github.com/roboticslab-uc3m/bullet-yarp-plugins: abandoned?
- https://github.com/roboticslab-uc3m/xgnitive: abandoned?
- https://github.com/roboticslab-uc3m/footsteps: abandoned?

As soon as YARP 3.5 is released, I'm going to gradually drop support for YARP 3.3 and require 3.4, introducing logging components in our repositories. For now, the component-less scheme I followed mostly everywhere is:

```cmake
yInfo() << "Bla bla bla" << value << "bla bla bla";
```

- A single sentence.
- Begins with uppercase letter.
- No ending period.

closed 2021-05-10 by PeterBowman

PeterBowman, 2021-07-19:
> As soon as YARP 3.5 is released, I'm going to gradually drop support for YARP 3.3 and require 3.4, introducing logging components in our repositories.

Done. Component names start with `rl` and reflect the main class of the library (or the library name itself), device or program, e.g. `rl.YarpCloudUtils`, `rl.KdlSolver`, `rl.DumpCanBus`. Components are declared and defined in their own .hpp/.cpp pair of files, then added to *CMakeLists.txt* ([example](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/e0ad59af44656723619aebe3442358c4d82a0667/libraries/YarpPlugins/KdlTreeSolver)). If only used in a single translation unit, they can be placed in an anonymous namespace and defined via `YARP_LOG_COMPONENT` ([example](https://github.com/roboticslab-uc3m/vision/blob/3e3551e96dc8e49d75a56b9ad01714e7064cd99c/programs/rgbDetection/RgbDetection.cpp#L25-L28)).