roboticslab-uc3m/questions-and-answers #91: Migrate to GitHub Actions

PeterBowman, 2020-12-06:
Travis CI is now making $$$ on "free" builds according to consumed job minutes ([ref](https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing)). Active users/orgs will hit the free plan limit sooner or later. It had happened to YARP folks and they decided to switch to GH Actions at https://github.com/robotology/yarp/issues/2410. Now, we have reached this limit five days ago, too, so no more Travis builds can start at the rl-uc3m org.

I'm afraid [it's time to say goodbye](https://youtu.be/4L_yCwFD6Jo) to Travis CI.

PeterBowman, 2021-01-30:
@jgvictores has found https://github.com/cyberbotics/webots-animation-action. GH Actions seems such a powerful tool!

PeterBowman, 2021-02-15:
Useful links:
* https://docs.github.com/en/actions
* https://docs.github.com/en/actions/reference
* https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions
* https://docs.github.com/en/actions/reference/environment-variables#default-environment-variables
* https://docs.github.com/en/actions/reference/specifications-for-github-hosted-runners#supported-software
* https://docs.github.com/en/actions/learn-github-actions
* https://docs.github.com/en/actions/learn-github-actions/migrating-from-travis-ci-to-github-actions

PeterBowman, 2021-02-15:
Enabled in color-debug: https://github.com/roboticslab-uc3m/color-debug/commit/76a82da9b5f900b1cd9e146fc863e96d5ce4230d.

PeterBowman, 2021-02-16:
Default CMake version in Ubuntu bionic and focal images is 3.19. YARP/YCM require at least 3.12. To stay on the safe side, I'm going to force builds against CMake 3.12 with https://github.com/jwlawson/actions-setup-cmake. Note conventions have changed and modern CMake projects can be configured, built and installed ([docs](https://cmake.org/cmake/help/latest/manual/cmake.1.html)) with:

```
cmake -S path_to_source -B path_to_build # creates directory structure if necessary
cmake --build path_to_build
cmake --install path_to_build
```

Pre-CMake 3.13 projects need to change the working directory and have the build dir created before first config ([docs](https://cmake.org/cmake/help/v3.12/manual/cmake.1.html)):

```
cmake -E make_directory path_to_build
# you can use `cd` here or prepend `cmake -E chdir` to each command as shown below:
cmake -E chdir path_to_build cmake ..
cmake -E chdir path_to_build cmake --build .
cmake -E chdir path_to_build cmake --build . --target install
```

See https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/780d25a92db5a1dc58b541bc3d470142f7874a8b.

PeterBowman, 2021-02-16:
Step commands ([`jobs.<job_id>.steps[*].run`](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#jobsjob_idstepsrun)) are executed in the default OS-shell (Windows: PowerShell, non-Windows: Bash) unless specified otherwise via [`jobs.<job_id>.steps[*].shell`](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#using-a-specific-shell). The `bash` option is also available on Windows systems using the bash shell included with Git for Windows. If all steps in one or all jobs are meant to be executed on both Windows and non-Windows systems, adding these lines instructs the runner to always prefer `bash` so that there is no need to add a `shell` property per step:

```yaml
defaults:
  run:
    shell: bash
```

See example at [color-debug](https://github.com/roboticslab-uc3m/color-debug/blob/ad177c4733a943d0594fffb9223fe0052a005060/.github/workflows/ci.yml). This default value may target all steps in one job ([docs](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#jobsjob_iddefaultsrun)) or all steps in all jobs ([docs](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#defaultsrun)).

I'm not sure how portable Bash really is, but it seems to work for our simple use-case in color-debug (Linux+Windows). An alternative, multi-platform strategy involves using CMake as a shell command launcher via `cmake -P {0}` ([ref](https://cristianadam.eu/20191222/using-github-actions-with-c-plus-plus-and-cmake/)). This solution has been adopted in YARP ([ref](https://github.com/robotology/yarp/blob/e1051ae266551ba7cccddb236ae7126b0a03e82f/.github/workflows/ci.yml#L854-L922)).

PeterBowman, 2021-02-16:
Commit https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/4cf5926b4a5172946cf9ff945779f63efba93cf5 introduces caching of main dependencies (YARP and KDL) with [ccache](https://software.intel.com/content/www/us/en/develop/articles/accelerating-compilation-part-1-ccache.html). Comparing build times:
- Not cached ([run](https://github.com/roboticslab-uc3m/kinematics-dynamics/actions/runs/572693479)): **~6 minutes** for YARP, ~1 minute for KDL.
- Cached ([run](https://github.com/roboticslab-uc3m/kinematics-dynamics/actions/runs/572720040)): **29 seconds** for YARP, 2 seconds for KDL.

I am using https://github.com/hendrikmuhs/ccache-action, which internally implements [this methodology](https://cristianadam.eu/20200113/speeding-up-c-plus-plus-github-actions-using-ccache/) using the official [cache action](https://github.com/actions/cache) ([guide](https://docs.github.com/en/actions/guides/caching-dependencies-to-speed-up-workflows)).

The following CMake options enable ccache for selected builds:

```
cmake .. -DCMAKE_C_COMPILER_LAUNCHER=ccache -DCMAKE_CXX_COMPILER_LAUNCHER=ccache
```

**Note**: this is a CMake variable, [`CMAKE_<LANG>_COMPILER_LAUNCHER`](https://cmake.org/cmake/help/v3.12/variable/CMAKE_LANG_COMPILER_LAUNCHER.html#variable:CMAKE_%3CLANG%3E_COMPILER_LAUNCHER). In CMake 3.17, an environment variable of the same name was introduced ([ref](https://cmake.org/cmake/help/latest/envvar/CMAKE_LANG_COMPILER_LAUNCHER.html)).

Speaking of kinematics-dynamics, I am enabling ccache only for YARP and KDL. Also, these builds are benefiting from parallelization via `cmake --build <dir> --parallel` (uses default parallel level if no value is provided). Intentionally avoiding parallel builds of YCM due to numerous failures experienced in past Travis jobs. I am not boosting the main build (i.e. kinematics-dynamics) on purpose, might change my mind later on.

PeterBowman, 2021-02-17:
It turns out scheduling jobs the way we did with Travis is a but more clumsy on GHA. It is not possible to evaluate conditions when defining an inclusion in the job matrix (if *event_type* is  *schedule* then add job). All combinations for scheduled and non-scheduled event types must be defined at once, then we'd just exclude YARP master branch combinations for non-scheduled jobs. However, [`jobs.<job_id>.if`](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#jobsjob_idif) cannot be used here since the matrix is defined after evaluating the condition ([ref](https://github.community/t/matrix-cannot-be-used-in-jobs-level-if/17177), https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/ebddfcc131056568a7ae7b8656f92564b2c85ee7).

A dummy matrix key that depends on the event type can be used to determine exclusions, though, as explained [here](https://github.community/t/how-to-conditionally-include-exclude-items-in-matrix-eg-based-on-branch/16853/6): https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/038a706bd4a03895da68cd9691850dca8b88afed. Note it is not possible to apply a similar trick for conditional job inclusions: https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/fa3d1f579824472f328e3f6bf62746f4bc0f6143.

Alternatives involve using a dynamic matrix which is conditionally defined in one job and actually consumed in another, dependent job: [ref](https://github.community/t/how-to-conditionally-include-exclude-items-in-matrix-eg-based-on-branch/16853/7). IMO even more clumsy. Of course I could just avoid all this mess and **always** run jobs on YARP's master branch, i.e. on every push, and possibly set [`jobs.<job_id>.continue-on-error`](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#jobsjob_idcontinue-on-error) to `true` so that the workflow run passes regardless of errors (which I hope would be notified somehow).

PeterBowman, 2021-02-17:
The amor-api private repo now participates in the CI workflow at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/201fd98fd4a1132633aecbea93596cd504e73f38 (WIP). In order to grant access, I am providing my personal access token (PAT) via the org-wide secret `AMOR_API`, currently enabled only on repos that need to fetch amor-api sources (kin-dyn, yarp-devices and amor-main). More info:

* https://docs.github.com/en/actions/reference/encrypted-secrets
* https://docs.github.com/en/actions/reference/authentication-in-a-workflow
* https://docs.github.com/en/github/authenticating-to-github/creating-a-personal-access-token
* https://github.community/t/best-way-to-clone-a-private-repo-during-script-run-of-private-github-action/16116/7

PeterBowman, 2021-02-17:
Similarly to Travis CI, GitHub Actions workflow runs can be skipped if the commit description includes a `[skip ci]` string.

Quoting https://github.blog/changelog/2021-02-08-github-actions-skip-pull-request-and-push-workflows-with-skip-ci/:

> GitHub Actions now supports skipping `push` and `pull_request` workflows by looking for some common keywords in your commit message.
>
>If any commit message in your push or the HEAD commit of your PR contains the strings `[skip ci]`, `[ci skip]`, `[no ci]`, `[skip actions]`, or `[actions skip]` workflows triggered on the `push` or `pull_request` events will be skipped.

PeterBowman, 2021-02-17:
> Speaking of kinematics-dynamics, I am enabling ccache only for YARP and KDL. Also, these builds are benefiting from parallelization via `cmake --build <dir> --parallel` (uses default parallel level if no value is provided). (...) I am not boosting the main build (i.e. kinematics-dynamics) on purpose, might change my mind later on.

Since ccache seems reliable enough, I have enabled caching for the main build, too. Not using `--parallel` for better browsing of the compile log.

PeterBowman, 2021-02-18:
> Alternatives involve using a dynamic matrix which is conditionally defined in one job and actually consumed in another, dependent job: [...]

I am not the only one who seeks this feature: [community post](https://github.community/t/continue-on-error-allow-failure-ui-indication/16773), https://github.com/actions/toolkit/issues/399. The `continue-on-error` seems to prevent a specific job from canceling other jobs, but still marks the whole run as failed. I was expecting some sort of notification about allowed failures as other users report in the linked issue (this was a Travis feature). In contrast, `fail-fast` defines the default behavior for all jobs, which is basically the same thing as `continue-on-error` with a different scope. See also https://github.com/actions/toolkit/issues/399#issuecomment-674618750.

Why would we want this? For early testing of experimental upstream dependencies (i.e. YARP/YCM). Options:
* (current *status quo*) Build pushes and pull requests against stable robotology branches (8 jobs AToW). On schedule (once per week), trigger an automated build and add more jobs targeting unstable (master) YARP/YCM branches (totalling 12 jobs). Who is notified when a scheduled run fails? According to [docs](https://docs.github.com/en/actions/guides/about-continuous-integration#notifications-for-workflow-runs):
  > Notifications for scheduled workflows are sent to the user who initially created the workflow. If a different user updates the cron syntax in the workflow file, subsequent notifications will be sent to that user instead. If a scheduled workflow is disabled and then re-enabled, notifications will be sent to the user who re-enabled the workflow rather than the user who last modified the cron syntax.
* Always build against stable and unstable robotology branches. In the current matrix configuration, this would result in 12 jobs per workflow run. Note GH Actions for free plans is limited to 20 concurrent jobs in the personal/organization account ([ref](https://docs.github.com/en/actions/reference/usage-limits-billing-and-administration#usage-limits)). I understand that if two runs are started on two repositories in our org, there will be 20-12+12=4 jobs awaiting the completion of a different run.

> A dummy matrix key that depends on the event type can be used to determine exclusions, though, as explained [here](https://github.community/t/how-to-conditionally-include-exclude-items-in-matrix-eg-based-on-branch/16853/6): [roboticslab-uc3m/kinematics-dynamics@038a706](https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/038a706bd4a03895da68cd9691850dca8b88afed).

This workaround doesn't cope well with JSON-based matrix entries: https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/5b2e3d8c9daa3b436aebf9c41de90a9f27b33ffe. The run got stuck and I had to cancel it. This is a rather lame reason in favor of switching to the second option (always 12 jobs), but still...

PeterBowman, 2021-02-18:
Commit https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/59c6b9a5c2d9e5cbfdb9d34c147f98de8fe0cd5c enables path exclusion on push events in such a way that a new run is not triggered whenever the algorithm detects that no file outside the `doc/` directory has been changed, also ignoring all `.md` files.

Docs:

* https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#onpushpull_requestpaths
* https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#filter-pattern-cheat-sheet

PeterBowman, 2021-02-19:
Mostly ready at kinematics-dynamics: [ci.yml](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/993e5520cae2538f420562490829eba6e5f271fe/.github/workflows/ci.yml). Cron jobs have been configured in a weekly fashion, on Sunday, 02:00 UTC. Also, I have prepared a template for quick reuse in other repos: [roboticslab-uc3m/.github](https://github.com/roboticslab-uc3m/.github/blob/9aecb08fec33d208aed6d02a63fdc6d2a155a99a/workflow-templates/ci.yml), [docs](https://docs.github.com/en/actions/learn-github-actions/sharing-workflows-with-your-organization#sharing-secrets-within-an-organization).

PeterBowman, 2021-02-23:
TO-DO list (from https://travis-ci.com/github/roboticslab-uc3m):
- [ ] [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech)

PeterBowman, 2021-02-27:
We could also migrate the generation of Doxygen documentation and GitBook manuals (currently at robots.uc3m.es) to GH Actions. For instance, [this script](https://github.com/robotology/gazebo-yarp-plugins/blob/0ea3ed6f190c2274865160961ab5e36133a8106a/.github/workflows/gh-pages.yml) was used in robotology/gazebo-yarp-plugins to generate http://robotology.github.io/gazebo-yarp-plugins/master/.

Links:
* https://docs.github.com/en/github/working-with-github-pages
* https://github.com/marketplace?query=github+pages
* https://github.com/jgvictores/wikirobots/blob/master/scripts/update-daily.sh ([permalink](https://github.com/jgvictores/wikirobots/blob/2eabfce68b159324eb292caf2aadb1abc7902bde/scripts/update-daily.sh))

PeterBowman, 2021-03-01:
Three repositories are currently configured to run weekly CI builds: [tools](https://github.com/roboticslab-uc3m/tools/actions/runs/606839481), [speech](https://github.com/roboticslab-uc3m/speech/actions/runs/606893870), [kin-dyn](https://github.com/roboticslab-uc3m/kinematics-dynamics/actions/runs/606813011).

Remarks:
* Scheduled runs are not timely. Individual jobs may start even half an hour after the appointed time, also they don't seem to be started all at once.
* GitHub removes cached files if not used after 7 days ([ref](https://docs.github.com/en/actions/guides/caching-dependencies-to-speed-up-workflows#usage-limits-and-eviction-policy)). I wonder how precise/strict is this. I have found that kin-dyn cache was restored after one week plus 3 minutes: [ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/runs/1996430412?check_suite_focus=true#step:9:9) (enable "Show timestamps").

PeterBowman, 2021-03-07:
This is nice: the .md filter prevented https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/a9acac8b8246c77be885f83648b12eddcac43d3c from triggering a CI run, but the weekly cron workflow started nonetheless ([run](https://github.com/roboticslab-uc3m/kinematics-dynamics/actions/runs/628562910)).

I see that cache eviction is not happening on weekly crons. I think it's safe to keep on migrating to GHA on the remaining repos. However, I'm probably dilating the no-build period on crons for selected, low-traffic repositories listed at https://github.com/roboticslab-uc3m/questions-and-answers/issues/91#issuecomment-784199905.

PeterBowman, 2021-03-11:
I have integrated https://github.com/fkirc/skip-duplicate-actions in our CI workflow: https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/55cfdcc78f53b71da69f746cfa84b15e6483a2ea. It covers the following use cases:

* Skip duplicate workflow-runs: don't trigger a run twice if it had been already completed on a feature branch before merging. For instance, let's assume we pushed and run CI of commit X on branch Y, then Y is merged into master to produce commit Z. CI jobs are skipped if Z was either a fast-forward commit, or a merge commit on top of master. If branch Y has outdated commits (i.e. some changes were pushed to master before the final merge of Y), then the run for Z is triggered as expected.
* Cancel outdated workflow-runs: cancel an ongoing run if more changes have been pushed to the same branch.

I believe these features will be eventually integrated into vanilla GHA. On the other hand, I don't quite understand "Skip ignored paths". I think GHA already provides backtracking, or at least the following scenario has been successfully tested:

* Start branch *my_feature*, commit a change in C++ code, push and await CI completion.
* Commit more changes to C++ code, but don't push yet.
* Commit a change to any .md file, commit and push. Note `**.md` was configured in `paths-ignore`.

I thought GHA would skip the workflow run for the last action because of "stupidly looking at the current commit", as stated in the README for skip-duplicate-actions. It didn't do so, though. I think it's performing a two-way comparison on each push: [ref](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#git-diff-comparisons).

PeterBowman, 2021-03-12:
> We could also migrate the generation of Doxygen documentation (...)

Done in kin-dyn at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/aed50de49991aceae784fb09f9ef3bd043623742 using https://github.com/crazy-max/ghaction-github-pages ([doxygen.yml](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/63871757390c48ab6f33b77a902e09a0d914e25c/.github/workflows/doxygen.yml), https://roboticslab-uc3m.github.io/kinematics-dynamics). I was considering two ways to do this:

* Add a new job to the CI workflow, conditionally enabled in master branch. In short: triggered on every push to master.
* Add a new workflow that would trigger either on every push to master, or periodically via cron.

I've chosen the latter and triggering on every push as long as GH detects changes on selected directories. I am also enabling manual dispatch, i.e. new runs can be manually triggered via GUI (see "Run workflow" button [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/actions/workflows/doxygen.yml)).

Notes:

* Nuked main.dox in favor of README.md as the main page for doxy.
* I resorted to using absolute paths pointing at GH CDN for images to work both in GH markdown and in Doxygen markdown: https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/63871757390c48ab6f33b77a902e09a0d914e25c.
* We started generating our Doxyfiles via CMake thanks to YCM at https://github.com/roboticslab-uc3m/questions-and-answers/issues/79. This decision was tailored to our robots.uc3m.es server because of working with obsolete packages (Ubuntu 15.10), therefore YCM allowed us to build and use CMake 3.5, yet take advantage of modern [FindDoxygen](https://cmake.org/cmake/help/v3.9/module/FindDoxygen.html) functions introduced in CMake 3.9. Note recent releases of Doxygen actually require a fix introduced in CMake 3.13 (also handled by YCM: https://github.com/robotology/ycm/pull/249). Since we are definitelly dropping robots.uc3m.es for documentation generation, I'm raising the requirements on doc/CMakeLists.txt to CMake 3.9 for stand-alone doxy builds ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/63871757390c48ab6f33b77a902e09a0d914e25c/doc/CMakeLists.txt#L5)).
* Had to tweak a few Doxygen options, see https://www.doxygen.nl/manual/config.html.

PeterBowman, 2021-03-23:
> We could also migrate the generation of (...) GitBook manuals (...)

Done at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/c9a99f03dcace6681d8c0d17fff6b33ab54b2dfc. Initially, I resorted to installing Calibre and GitBook from scratch (node modules + plugins), which took +3 minutes to complete: [gitbook.yml](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/c37d8a6bf09a27bb970d9b452d97ba9f3418aaec/.github/workflows/gitbook.yml). To avoid this, I tried to take advantage of custom Docker containers, thus cutting run times down to one minute (but no PDF). Links:

- https://github.com/billryan/docker-gitbook ([DockerHub](https://hub.docker.com/r/billryan/gitbook), [MicroBadger](https://microbadger.com/images/billryan/gitbook))
- https://hub.docker.com/_/node (using `node:10.24-alpine`)
- https://calibre-ebook.com/download_linux
- https://docs.docker.com/develop/develop-images/multistage-build/
- https://docs.docker.com/develop/develop-images/dockerfile_best-practices/
- https://docs.github.com/en/packages/guides/container-guides-for-github-packages
- https://github.com/roboticslab-uc3m/docker-gitbook
- https://github.com/orgs/roboticslab-uc3m/packages/container/package/gitbook

Sadly, PDF generation is broken, so we've decided to take it out (along with MOBI and EPUB files). After installing hundreds of MBs of dependencies (Calibre/ebook-convert + svgexport/puppeteer/Chromium) and solving a known `--no-sandbox` issue (see [intructions](https://github.com/puppeteer/puppeteer/blob/main/docs/troubleshooting.md#running-puppeteer-in-docker), [permalink](https://github.com/puppeteer/puppeteer/blob/ff860819a75fd784555ad5d3d5a5458263d3ef4e/docs/troubleshooting.md#running-puppeteer-in-docker)), it gets stuck in the `gitbook pdf` step (but no errors are thrown). This is the closest I have gotten, aditionally it is necessary to create a new user and assign permissions (see linked puppeteer's troubleshooting guide):

```dockerfile
FROM alpine:latest AS download_calibre
ENV CALIBRE_DOWNLOAD_URL https://download.calibre-ebook.com/5.13.0/calibre-5.13.0-x86_64.txz
RUN mkdir /tmp/calibre && \
    wget --no-check-certificate -qO- $CALIBRE_DOWNLOAD_URL | tar xvJ -C /tmp/calibre

FROM node:10.24-buster-slim
ENV LD_LIBRARY_PATH $LD_LIBRARY_PATH:/opt/calibre/lib
ENV PATH $PATH:/opt/calibre/bin
COPY --from=download_calibre /tmp/calibre /opt/calibre
RUN npm install -g --unsafe-perm \
        gitbook-cli \
        gitbook-summary \
        svgexport && \
    gitbook install && \
    rm -rf /tmp/* && \
    npm cache clean --force
EXPOSE 4000
VOLUME /gitbook
WORKDIR /gitbook
CMD ["gitbook", "--help"]
```

TODO per [update-daily.sh](https://github.com/jgvictores/wikirobots/blob/2eabfce68b159324eb292caf2aadb1abc7902bde/scripts/update-daily.sh):

- [x] [teo-developer-manual](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/af9b45123f782b293059e1cd35a979d93ab8422c/scripts/admin/update-gitbook.sh)
- [ ] [developer-manual](https://github.com/roboticslab-uc3m/developer-manual/blob/3d100503463748491b7a8e730a9487953aebf5d0/scripts/admin/update-gitbook.sh)
- [ ] [installation-guides](https://github.com/roboticslab-uc3m/installation-guides/tree/0492106f1d967b172958579340cec5c7fefa4a9d/scripts/admin)

jgvictores, 2021-04-16:
Note: https://github.com/asrob-uc3m/repostatistics -> http://robots.uc3m.es/contributors/

closed 2021-05-09 by PeterBowman

jgvictores, 2021-05-10:
@PeterBowman THANK YOU SO MUCH!!!! 🎉🎉🎉🎉🎉

PeterBowman, 2021-06-01:
I have just learned that "scheduled workflows are automatically disabled when no repository activity has occurred in 60 days" ([ref](https://docs.github.com/en/actions/managing-workflow-runs/disabling-and-enabling-a-workflow)). The following notice is displayed right now in the [Action's tab of the speech repo](https://github.com/roboticslab-uc3m/speech/actions/workflows/ci.yml):

[image]

If we didn't care about the future of the planet, then we could schedule a cron on our robots.uc3m.es server to run `gh workflow enable ci` every now and then, see [docs](https://docs.github.com/en/actions/managing-workflow-runs/disabling-and-enabling-a-workflow#disabling-and-enabling-workflows-with-github-cli).

PeterBowman, 2021-06-03:
Note sometimes builds may fail for no apparent reason. We have recently seen pretty weird segfaults on unit tests affecting just Clang builds on YARP master and Ubuntu bionic, among other issues. It is fine, probably cache artifacts are conflicting on linkage or something like that. Don't panic and invalidate the cache like this: https://github.com/roboticslab-uc3m/tools/commit/38b02d2b807e91a4009b460ac6973289da6cc004. Make sure to also target master builds, then manually trigger the workflow again to make sure it's fixed and rebuild the cache.