roboticslab-uc3m/questions-and-answers #92: Add C++ examples to main build

PeterBowman, 2020-12-17:
Our vision repo has recently gained Python bindings ([ref](https://github.com/roboticslab-uc3m/vision/tree/b30fae4b173d638a430bc4bbf45d188fcc4e5117/bindings)) that get compiled along with everything else, that is, they generate targets in the main build (although disabled by default). So far we excluded them from `make -C build/ all` and forced users to generate a `bindings/build/` directory, configure CMake and build separately (of course, the main project ought to be compiled and installed at this point).

Why not applying this very same logic to C++ examples? I have given it a try at https://github.com/roboticslab-uc3m/vision/commit/182286aeae2f96501954754b243ad5cb11f7622f with the following changes:

* created [alias library targets](https://cmake.org/cmake/help/latest/command/add_library.html#alias-libraries) that add the `ROBOTICSLAB::` prefix
 * new `add_subdirectory(examples/cpp)` at the main *CMakeLists.txt*
 * moar `add_subdirectory()` within `examples/cpp/CMakeLists.txt` to traverse examples and a default-OFF cache option
 * prevent examples from being installed
 * conditional checks for `find_package(ROBOTICSLAB_VISION REQUIRED)`
 * removed lines related to these examples from *travis.yml*, now passing `ENABLE_examples=ON` (https://github.com/roboticslab-uc3m/questions-and-answers/issues/42)

I found it works out-of-the-box for whole-project builds with `ENABLE_examples=ON` as well as builds that target nothing more than one specific example.

Drawbacks, i.e. what we actually have to or should alter in the *CMakeLists.txt* for each example:

* Required if applicable:
  ```cmake
  if(NOT TARGET ROBOTICSLAB::VisionInterfaces)
      find_package(ROBOTICSLAB_VISION REQUIRED)
  endif()
  ```
* Several identical `Found YARP: <path> (<version>)` lines in the CMake configure output originating from each example *CMakeLists.txt* caused by `find_package(YARP)`. This can be easily solved with a `QUIET` option, ideally set from parent list file:
  ```cmake
  find_package(YARP 3.3 ${_example_main_quiet} REQUIRED) # variable expands either to nothing or `QUIET`
  ```

PeterBowman, 2020-12-18:
> This can be easily solved with a `QUIET` option, ideally set from parent list file:

I got this solved with a similar solution already adopted for `find_package(ROBOTICSLAB_VISION REQUIRED)`: https://github.com/roboticslab-uc3m/vision/commit/70af1cdef64643e008b108a385e08403d10793f5. It is important to remark, that...

...this is mandatory since calling `find_package` on the project currently being built causes trouble:

```cmake
if(NOT TARGET ROBOTICSLAB::YarpCloudUtils)
    find_package(ROBOTICSLAB_VISION REQUIRED)
endif()
```

....and this is not, but prevents from issuing duplicate `find_package` calls, and therefore extra CMake log:

```cmake
if(NOT YARP_FOUND)
    find_package(YARP 3.3 REQUIRED COMPONENTS os sig)
endif()
```

Regarding the main build, all dependencies (including 3rd-party ones) should be appended (as usual) to the "Soft dependencies" list at root *CMakeLists.txt*, for clarity and uniformity. Then, more conditionals could be added to *examples/cpp/CMakeLists.txt*:

```cmake
option(ENABLE_examples "Enable/disable C++ examples" OFF)

if(ENABLE_examples)
    if(TARGET ROBOTICSLAB::VisionInterfaces)
        add_subdirectory(exampleColorRegionDetector)
        add_subdirectory(exampleHaarDetector)
    endif()

    if(TARGET ROBOTICSLAB::YarpCloudUtils)
        add_subdirectory(exampleMeshFromCloud)
        add_subdirectory(exampleMeshFromLiveRGBD)
        add_subdirectory(exampleProcessCloud)
        add_subdirectory(exampleSceneReconstructionClient)
    endif()

    add_subdirectory(exampleRemoteGrabber)
    add_subdirectory(exampleRemoteRGBDSensor)
endif()
```

I mean: adding a `if(MY_DEPENDENCY_FOUND)` clause around the offending example subdirectory.

PeterBowman, 2020-12-18:
Proposal: whenever a library target alias is created, use it (https://github.com/roboticslab-uc3m/vision/commit/247265b7e6c0620833d51b328325f98da788b3b1). For consistency, I'd create such aliases for library created by the project, even if those are not linked by any internal component (example, unit test or application).

PeterBowman, 2020-12-18:
Once added to the main build, should C++ be enabled by default (currently `OFF`)?

jgvictores, 2020-12-18:
> Once added to the main build, should C++ be enabled by default (currently `OFF`)?

I'd say yep. ^^

jgvictores, 2020-12-18:
> Once added to the main build, should C++ be enabled by default (currently `OFF`)?

But would be okay with nay.

PeterBowman, 2020-12-18:
I believe we can stay with default `OFF` for now. TODO:

- [ ] [vision](https://github.com/roboticslab-uc3m/vision/tree/f8f40ee68a1a15038f818d8dda9d25d49f3108db/examples/cpp)
- [ ] [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices/tree/5a2abd14e71e64d6cef7260f56bfcfecb8ddafaf/examples/cpp)
- [ ] [speech](https://github.com/roboticslab-uc3m/speech/tree/41bf9787ab3a3b0d6c06159b4bbfcc2505bd0145/examples/cpp)
- [ ] [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/c19ccf27323448c8b65fc30b45f3c8e39aa2d078/examples/cpp)
- [ ] [tools](https://github.com/roboticslab-uc3m/tools/tree/93d7f44ded29c6573b6e5f6e554684975e007981/examples/cpp)
- [ ] [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/ce3771a9926db3c41294e740b216020479ce2a57/examples/cpp)

closed 2021-02-15 by PeterBowman

jgvictores, 2021-02-15:
Yay! @PeterBowman thanks!!