roboticslab-uc3m/questions-and-answers #96: sp_ineachdb and also sp_msforeachtable return only 50 records instead of hundreds

lapus014, 2023-04-18:
When i try to use sp_msforeachtable  or sp_ineachdb it returns only 50 (with sp_msforeachtable ) and 48 (with sp_ineachdb ) results, instead of hundreds of results that should b returned. I see the results per DB when execute the code per DB without using the above procedures.
But what can be the issue, i need to run the code for all dbs and not only specific one.
This is the code:

IF OBJECT_ID('tempdb..#identity_columns') IS NOT NULL DROP TABLE #identity_columns
GO

CREATE TABLE #identity_columns
(	[database_name] SYSNAME NOT NULL,
	[schema_name] SYSNAME NOT NULL,
	table_name SYSNAME NOT NULL,
	column_name SYSNAME NOT NULL,
	[type_name] SYSNAME NOT NULL,
	maximum_identity_value BIGINT NOT NULL,
	current_identity_value BIGINT NULL,
	percent_consumed DECIMAL(25,4) NULL	);
 
DECLARE @Table_Name NVARCHAR(MAX);
DECLARE @Schema_Name NVARCHAR(MAX)
DECLARE @command varchar(1000);
 
SELECT @command= '
INSERT INTO #identity_columns
	([database_name], [schema_name], table_name, column_name, [type_name], maximum_identity_value, current_identity_value)
SELECT
	DB_NAME() AS database_name,
	''' + schemas.name + ''' AS schema_name,
	''' + tables.name + ''' AS table_name,
	''' + columns.name + ''' AS column_name,
	''' + types.name + ''' AS type_name,
	CASE
		WHEN ''' + types.name + ''' = ''TINYINT'' THEN CAST(255 AS BIGINT)
		WHEN ''' + types.name + ''' = ''SMALLINT'' THEN CAST(32767 AS BIGINT)
		WHEN ''' + types.name + ''' = ''INT'' THEN CAST(2147483647 AS BIGINT)
		WHEN ''' + types.name + ''' = ''BIGINT'' THEN CAST(9223372036854775807 AS BIGINT)
		WHEN ''' + types.name + ''' IN (''DECIMAL'', ''NUMERIC'') THEN CAST(REPLICATE(9, (' + CAST(columns.precision AS VARCHAR(MAX)) + ' - ' + CAST(columns.scale AS VARCHAR(MAX)) + ')) AS BIGINT)
		ELSE -1
	END AS maximum_identity_value,
	IDENT_CURRENT(''[' + schemas.name + '].[' + tables.name + ']'') AS current_identity_value;
'
FROM sys.tables
INNER JOIN sys.columns
ON tables.object_id = columns.object_id
INNER JOIN sys.types
ON types.user_type_id = columns.user_type_id
INNER JOIN sys.schemas
ON schemas.schema_id = tables.schema_id
WHERE columns.is_identity = 1;
 
EXEC dbo.sp_ineachdb @command
UPDATE #identity_columns
	SET percent_consumed = CAST(CAST(current_identity_value AS DECIMAL(25,4)) / CAST(maximum_identity_value AS DECIMAL(25,4)) AS DECIMAL(25,2)) * 100;
 select * from #identity_columns order by percent_consumed desc

PeterBowman, 2023-04-18:
Sorry, your issue is out of the scope of this repo.

closed 2023-04-18 by PeterBowman

lapus014, 2023-04-18:
> Sorry, your issue is out of the scope of this repo.

What repo should I use?

PeterBowman, 2023-04-18:
> What repo should I use?

Google.